P1sec/pycrate #2: Dynamic compilation of an ASN.1 object

H21lab, 2017-11-16:
Is there a way of doing Dynamic ASN.1 compilation in pycrate in similar way as in libmich? Some example would be very useful. Thank you

mitshell, 2017-11-16:
No, this is not supported (yet). Things on the ASN.1 side changed quite a lot, compared to libmich. In particular, all objects compiled are kept in their module (e.g. in GLOBAL.MOD['SomeModule']).

I may be able to make a specific part / function in the ASN.1 runtime (that import the whole ASN.1 compiler) and compile some object defined within a Python string into an object into the GLOBAL.MOD['_USER_'] special module.
Would that help ?

H21lab, 2017-11-17:
Yes, I think this would help. To me this is useful for importing only selected objects from ASN.1 files instead of compiling the whole ASN.1 files. By this the light scripts could be created without too much dependencies.

mitshell, 2017-11-17:
As a workaround, you can also strip an ASN.1 definition with only the definitions you need, and compile it into a Python file. You will then just need to load this Python file to get the objects you want.

I have not much time yet to work on the ASN.1 compiler side, unfortunately. Hence, this issue may stay open for few months.

H21lab, 2017-11-17:
Thank you for the feedback. I have workaround it in this way.