P1sec/pycrate #6: Exceptions in from_ber decoding

H21lab, 2017-12-01:
Hi, 
by running the script [test_tcapmap.py](https://github.com/ANSSI-FR/pycrate/pull/5) the following exceptions has been raised. 
Maybe it is ok, but these exceptions does not seem to be ASN1Err exceptions. I do not include the input payload because these errors can be reproduced by running the script.

Thank you for your opinion on this

```
# decode using pycrate
tcap = TCAP_MAP.TCAP_MAP_Messages.TCAP_MAP_Message
tcap.from_ber(binascii.unhexlify(tcap_hex))
```

Exceptions:
```
bitlen overflow: 1600, max 32
Traceback (most recent call last):
  File "test_tcapmap.py", line 89, in <module>
    tcap.from_ber(binascii.unhexlify(tcap_hex))
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 856, in from_ber
    self._from_ber(char, TLV)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 842, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj_construct.py", line 506, in _decode_ber_cont
    Cho._from_ber(char, [tlv])
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 842, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj_construct.py", line 1397, in _decode_ber_cont
    Comp._from_ber(char, [tlv[ind]])
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 842, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj_construct.py", line 1397, in _decode_ber_cont
    Comp._from_ber(char, [tlv[ind]])
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 842, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj_construct.py", line 506, in _decode_ber_cont
    Cho._from_ber(char, [tlv])
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 842, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj_ext.py", line 441, in _decode_ber_cont
    self._val = ASN1CodecBER.scan_tlv_ws(char, tlv)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1570, in scan_tlv_ws
    return char.get_bytes(8*lval)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_core/charpy.py", line 376, in get_bytes
    .format(bitlen, self._len_bit-self._cur)))
CharpyErr: bitlen overflow: 1600, max 32
```

```
bytes buffer not long enough
Traceback (most recent call last):
  File "test_tcapmap.py", line 89, in <module>
    tcap.from_ber(binascii.unhexlify(tcap_hex))
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/asnobj.py", line 851, in from_ber
    TLV = [ASN1CodecBER.decode_single(char)[0]]
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1535, in decode_single
    V = cla.decode_all(char, lundef=False)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1557, in decode_all
    TLV, EOS = cla.decode_single(char, lundef)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1535, in decode_single
    V = cla.decode_all(char, lundef=False)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1557, in decode_all
    TLV, EOS = cla.decode_single(char, lundef)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1535, in decode_single
    V = cla.decode_all(char, lundef=False)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1557, in decode_all
    TLV, EOS = cla.decode_single(char, lundef)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1535, in decode_single
    V = cla.decode_all(char, lundef=False)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1557, in decode_all
    TLV, EOS = cla.decode_single(char, lundef)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1522, in decode_single
    cl, pc, tval = cla.decode_tag(char)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_asn1rt/codecs.py", line 1337, in decode_tag
    cl = char.get_uint(2)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_core/charpy.py", line 783, in get_uint
    return bytes_to_uint(self._buf[off_byte:1+off_byte+(bitlen>>3)], bitlen)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.2-py2.7.egg/pycrate_core/utils_py2.py", line 1301, in bytes_to_uint
    raise(PycrateErr('bytes buffer not long enough'))
PycrateErr: bytes buffer not long enough
```

mitshell, 2017-12-05:
Actually, everything that raises a PycrateErr, or an error inherited from PycrateErr, is kind of expected. In the ASN.1 runtime, I don't catch all and any Pycrate exceptions to then re-raise them with a specific ASN1Err: this is painful, and finally kind of useless. If you get an exception that is not from the pycrate library, this is an unexpected case that would need specific attention.
Thank you for your work.

H21lab, 2017-12-06:
Then I think it looks ok and lets close the issue. Thank you.

closed 2017-12-06 by H21lab