P1sec/pycrate #8: S1AP S1SetuResponse ServedGUMMEIsItem

GinesGarcia, 2018-01-23:
Hi,
I'm trying to create a S1SetupResponse that must contain ServedGUMMEIs and RelativeMMECapacity as protocolIE fields but, after running:
  - `PDU = S1AP.S1AP_PDU_Descriptions.S1AP_PDU`
  - `PP.pprint(PDU._cont['successfulOutcome']._cont['value']._get_const_tr()['S1SetupResponse'])`
     `<SuccessfulOutcome ([S1SetupResponse] SEQUENCE)>`
I tried to store both ServedGUMMEIs and RelativeMMECapacity in the PDU obj as follows:
`IEs.append({
					'id': 105, 
					'criticality': 'reject', 
					'value': ('servedGUMMEIs', {
						'servedPLMNs': {'PLMNidentity': b'\x00\x01\xf1'}, 
						'servedGroupIDs': {'MME-Group-ID': b'\x00\x04'}, 
						'servedMMECs': {'MME-Code': b'\x01'}}
					)
				})`

`IEs.append({'id': 87, 'criticality': 'ignore', 'value': ('RelativeMMECapacity', 10)})`
`val = ('successfulOutcome', {'procedureCode': 17, 'value': ('S1SetupResponse', {'protocolIEs': IEs}), 'criticality': 'ignore'})`
`PDU.set_val(val)`

 and I got this error message:
`S1SetupResponse.protocolIEs._item_.value: invalid object reference, 'servedGUMMEIs'`

It seems that those items are not implemented. If this is true, is there any way to add it?

Thanks in advance!

mitshell, 2018-01-23:
Everything from the S1AP ASN.1 specification is supported.

See here for the code in the corenet part, which does exactly that:
https://github.com/ANSSI-FR/pycrate/blob/master/pycrate_corenet/ProcCNS1ap.py#L1807

I can provide this short Python code to give a simple one-shot example:
```
from pycrate_core.utils import uint_to_bytes
from pycrate_mobile.TS24008_IE import PLMN
from pycrate_asn1dir import S1AP
from binascii import hexlify

PDU = S1AP.S1AP_PDU_Descriptions.S1AP_PDU

MME_Name     = 'TestMME'
MME_Capacity = 10 # uint8
MME_PLMN     = '00101' # 5 or 6-digits str
MME_GID      = 1 # uint16
MME_CODE     = 1 # uint8

ServedGUMMEIs = [
    {
     'servedPLMNs'   : [PLMN(val=MME_PLMN).to_bytes(), ], # add more PLMN in this 1st GUMMEI if needed
     'servedGroupIDs': [uint_to_bytes(MME_GID, 16), ], # add more MME Group ID in this 1st GUMMEI if needed
     'servedMMECs'   : [uint_to_bytes(MME_CODE, 8), ], # add more MME Code in this 1st GUMMEI if needed
     }, # add more GUMMEI dict if needed
    ]

S1SetupIEs = [
    {'id': 61,  'criticality': 'ignore', 'value': ('MMEname', MME_Name)},
    {'id': 105, 'criticality': 'reject', 'value': ('ServedGUMMEIs', ServedGUMMEIs)},
    {'id': 87,  'criticality': 'reject', 'value': ('RelativeMMECapacity', MME_Capacity)}, # add more optional IEs if needed
    ]

S1SetupVal = ('successfulOutcome', {
    'procedureCode': 17,
    'criticality': 'ignore',
    'value': ('S1SetupResponse', {'protocolIEs': S1SetupIEs})
    })

PDU.set_val( S1SetupVal )
print(hexlify(PDU.to_aper()).decode('ascii'))
```

closed 2018-01-23 by mitshell

GinesGarcia, 2018-01-24:
Thanks a lot for your help!