P1sec/pycrate #10: [NAS] Building EMMSecurityModeCommand message

GinesGarcia, 2018-03-22:
Hi,
I have been trying to create a EMMSecurityModeCommand(). I read about NAS in the documentation but I cannot find a clean way to do it. I did it by using the from_bytes() function and also creating an empty EMMSecurityModeCommand Object and setting its parameters but I couldn't, for example, to pass all the information when creating the object.

Is there any way to do it?

Thanks a lot for your help :)

mitshell, 2018-03-22:
Yes, like this:
    
    pdu = EMMSecurityModeCommand(val={'NASSecAlgo': b'\x11', 'NAS_KSI':2, 'UESecCap':b'\xe0\x60', 'IMEISVReq': 1})
    
You can also pass specific NASSecAlgo and UESecCap values instead of the raw buffer:
    
    pdu = EMMSecurityModeCommand(val={'NASSecAlgo':{'CiphAlgo':1, 'IntegAlgo':1}, 'NAS_KSI':2, 'UESecCap':{'EEA0': 1, 'EEA1_128': 1, 'EEA2_128': 1, 'EIA1_128': 1, 'EIA2_128': 1}, 'IMEISVReq': 1})

closed 2018-03-22 by mitshell

GinesGarcia, 2018-03-22:
Thank you very much for your help!!
this is definitely what I was looking for :)

Regards,
Ginés.