P1sec/pycrate #12: ESM info response is not decoding correctly

jothikumarmani, 2018-04-03:
Hi,
i was trying to execute the test folder examples found that mobile NAS message (ESM info Response : 0202da2807066f72616e6765) throwing error.

mitshell, 2018-04-03:
On my side, this is working fine:

    In [1]: from pycrate_mobile.NAS import *

    In [2]: parse_NAS_MO(unhexlify('0202da2807066f72616e6765'))
    Out[2]: 
    (<ESMInformationResponse : <ESMHeader : <EPSBearerId : 0><ProtDisc : 2><PTI : 2><Type : 218 (ESM information response)>><APN : <T : 40><L : 7><APN : <APNItem : <Len : 6><Value : b'orange'>>>><ProtConfig [transparent] : <T : 39><L : 1><V : 0x80>><ExtProtConfig [transparent] : <T : 123><L : 1><V : 0x00>>>,
     0)

    In [3]: show(_[0])
    ### ESMInformationResponse ###
     ### ESMHeader ###
      <EPSBearerId : 0>
      <ProtDisc : 2>
      <PTI : 2>
      <Type : 218 (ESM information response)>
     ### APN ###
      <T : 40>
      <L : 7>
      ### APN ###
       ### APNItem ###
        <Len : 6>
        <Value : b'orange'>

Could you give details on what you did, what Python version you are using, ... ?

jothikumarmani, 2018-04-03:
thanks for your prompt response, sorry i was doing some mistake on my setup(pycharm),and sorry for the inconvenience

closed 2018-04-03 by mitshell

jothikumarmani, 2018-04-04:
during the APN decoding, i was getting the below error so thought could be something wrong.
"APN, _from_char: uncorrect decoding for IE, APN", Anyway thanks for the support.