P1sec/pycrate #17: class ASN1Obj(Element) attributes

p1-scg, 2018-04-27:
Hello,

I have a question about implementation of attributes in class ASN1Obj:
 ``` 
    _name    = ''
    _mode    = MODE_TYPE
    _tag     = None
    _typeref = None
    _tr      = None
    _param   = False
    _parent  = None
    _opt     = False
    _def     = None
    _uniq    = False
    _group   = None
    _cont    = None
    _root    = None
    _ext     = None
    _val     = None
```
and I didn't catch up why there is in the Class, so it is a static value.
Especially, when you use decode an object with method from_ber(), and then you
decode another object with the same method (but empty value). The _val
variable will keep the value of the first object.
What is the reason of the class-level implementation vs the obj-level implementation?

Thanks,

mitshell, 2018-04-27:
Class attributes provide default values for all attributes.
Instance attributes provide, when defined, specific values that masquerade those from the class.
This saves some memory.

p1-scg, 2018-05-03:
Thank you for the explanation, I close the issue.

closed 2018-05-03 by p1-scg