P1sec/pycrate #18: [Malformed packet] SEQ and SEQ_OF object

p1-scg, 2018-05-04:
Hello,

I really enjoy using pycrate, to craft MAP messages and I just discover a bug and share
to you.

I send two MAP messages using InsertSubscriberDataArg.
The first test, I add one SEQ object in the gprsDataList SEQ_OF object, encode in
the InsertSubscriberDataArg object, and send the packet in the network, and it works well. 

In the second test, I add two SEQ objects in the grpsDataList SEQ_OF object,
and the packet is malformed according to Wireshark.

I also add more than two SEQ objects in the gprsDataList SEQ_OF object, and it
also fails.

TEST PCAP files
[pcap.tar.gz](https://github.com/ANSSI-FR/pycrate/files/1975331/pcap.tar.gz)

**FIRST TEST:**
In [1]: from pycrate_asn1dir import TCAP_MAP

In [2]: m = TCAP_MAP.MAP_MS_DataTypes.InsertSubscriberDataArg 

In [3]: m.set_val({u'gprsSubscriptionData': {u'gprsDataList': [{u'pdp-ContextId': 5, u'apn': '\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05', u'pdp-Type': '\xf1!', u'ext-QoS-Subscribed': '\x02k\x96\xfe\xfet\x02\x00\x00', u'ext2-QoS-Subscribed': '\x00\x8e\x00', u'qos-Subscribed': '\x13\x92\x1f'}]}})
In [4]: m.to_ber()
Out[4]: '04\xb02\xa100.\x02\x01\x05\x90\x02\xf1!\x92\x03\x13\x92\x1f\x94\x10\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05\x80\t\x02k\x96\xfe\xfet\x02\x00\x00\x82\x03\x00\x8e\x00'

In [5]: import binascii

In [6]: binascii.hexlify(m.to_ber())
Out[6]: '3034b032a130302e0201059002f121920313921f9410075005070707075005070707075005058009026b96fefe740200008203008e00'

**SECOND TEST:**
In [1]: from pycrate_asn1dir import TCAP_MAP
In [2]: m = TCAP_MAP.MAP_MS_DataTypes.InsertSubscriberDataArg

In [3]: m.set_val({u'gprsSubscriptionData': {u'gprsDataList': [{u'pdp-ContextId': 5, u'apn': '\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05', u'pdp-Type': '\xf1!', u'ext-QoS-Subscribed': '\x02k\x96\xfe\xfet\x02\x00\x00', u'ext2-QoS-Subscribed': '\x00\x8e\x00', u'qos-Subscribed': '\x13\x92\x1f'},{u'pdp-ContextId': 4, u'apn': '\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05', u'pdp-Type': '\xf1!', u'ext-QoS-Subscribed': '\x02k\x96\xfe\xfet\x02\x00\x00', u'ext2-QoS-Subscribed': '\x00\x8e\x00', u'qos-Subscribed': '\x13\x92\x1f'}]}})

In [4]: m.to_ber()
Out[4]: '0d\xb0b\xa1`0.\x02\x01\x05\x90\x02\xf1!\x92\x03\x13\x92\x1f\x94\x10\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05\x80\t\x02k\x96\xfe\xfet\x02\x00\x00\x82\x03\x00\x8e\x000.\x02\x01\x04\x90\x02\xf1!\x92\x03\x13\x92\x1f\x94\x10\x07P\x05\x07\x07\x07\x07P\x05\x07\x07\x07\x07P\x05\x05\x80\t\x02k\x96\xfe\xfet\x02\x00\x00\x82\x03\x00\x8e\x00'

In [5]: import binascii

In [6]: binascii.hexlify(m.to_ber())
Out[6]: '3064b062a160302e0201059002f121920313921f9410075005070707075005070707075005058009026b96fefe740200008203008e00302e0201049002f121920313921f9410075005070707075005070707075005058009026b96fefe740200008203008e00'

Thank you for the clarification.

mitshell, 2018-05-14:
I checked your issue:

1) The encoding of TCAP_MAP.MAP_MS_DataTypes.InsertSubscriberDataArg seems fine, also with several gprsDataList. You can use .to_ber_ws() and check the ._struct attribute generated. All SEQUENCE OF components seems encoded and decoded correctly on my side.

2) In the pcap trace you sent, I can decode correctly the 1st TCAP-MAP message, but not the 2nd one, where I get the following error:
```
ASN1BERDecodeErr: TCAP-MAP-Message.begin: missing mandatory component, otid
```
The otid field is early in the TCAP header, and not part of the MAP message. I guess you screwed something when generating the TCAP-MAP-Message buffer in the 2nd message. You should investigate in this way, and not about a potential issue in the pycrate BER encoder.

3) When you send a pcap, please keep only the packets required for the analysis and remove everything else (including the ssh session...).

Benoit

p1-scg, 2018-06-05:
Hello,
It seems to work if we use Pycrate TCAP-MAP message.
I miss out something when I manually encode TCAP layer.
Thank you for your competent advice.
So I close the issue

closed 2018-06-05 by p1-scg