P1sec/pycrate #19: [ASN1] Error while parsing open type value with CHOICE

Patrickx93, 2018-05-11:
```
errorIndication F1AP-PDU ::= initiatingMessage :
{
    procedureCode id-ErrorIndication,
    criticality ignore,
    value ErrorIndication : {
      protocolIEs{
        {
          id id-Cause,
          criticality ignore,
          value Cause : protocol : abstract-syntax-error-reject
        }
      }
    }
}
```
It looks like another colon from choice syntax is the problem, since all other open type value types worked correctly.

Here's the traceback:

```
Traceback (most recent call last):
  File "pycrate_asn1compile.py", line 145, in <module>
    sys.exit(main())
  File "pycrate_asn1compile.py", line 133, in main
    compile_text(''.join(txt), **ckw)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/proc.py", line 350, in compile_text
    compile_modules(remain)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/proc.py", line 884, in compile_modules
    ObjNew = asnobj_compile(Obj)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/proc.py", line 831, in asnobj_compile
    rest = Obj.parse_value(Obj._text_def)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5176, in _parse_value_choice
    text = ObjProxy.parse_value(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5251, in _parse_value_seq
    restval = ObjProxy.parse_value(value)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5459, in _parse_value_open
    textval = Obj.parse_value(textval)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5251, in _parse_value_seq
    restval = ObjProxy.parse_value(value)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5385, in _parse_value_seqof
    restval = ObjProxy.parse_value(value)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5251, in _parse_value_seq
    restval = ObjProxy.parse_value(value)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 4537, in parse_value
    return getattr(self, self._PARSE_VALUE_DISPATCH[self._type])(text)
  File "/usr/local/lib/python2.7/dist-packages/pycrate-0.3-py2.7.egg/pycrate_asn1c/asnobj.py", line 5432, in _parse_value_open
    .format(self.fullname(), text)))
pycrate_asn1c.err.ASN1ProcTextErr: value: invalid OPEN value, Cause : protocol : abstract-syntax-error-reject
```

mitshell, 2018-05-14:
Thanks for reporting

I checked with the following test case:
```
Test DEFINITIONS AUTOMATIC TAGS ::=

BEGIN

    LUT ::= CLASS {
        &id INTEGER UNIQUE,
        &Type
        } WITH SYNTAX {
        ID &id TYPE &Type
        }
    
    F0 ::= INTEGER
    F1 ::= BIT STRING
    F2 ::= CHOICE {
        c0 F0,
        c1 F1,
        c2 CHOICE {
            c20 F0,
            c21 F1
           }
        }
    
    Lut LUT ::= {
        { ID 0 TYPE F0 } |
        { ID 1 TYPE F1 } |
        { ID 2 TYPE F2 }
        }
    
    Struct ::= SEQUENCE {
        id   LUT.&id   ({Lut}),
        type LUT.&Type ({Lut}{@id})
        }
    
    test0 Struct ::= {
        id 0,
        type F0 : 1
        }
    
    test1 Struct ::= {
        id 2,
        type F2: c2: c20: 1
        }
END
```

The compiler fails to compile _test1_. Disabling the check here solves the issue:
https://github.com/ANSSI-FR/pycrate/blob/fdfd19633664644fd8aa94ae20eeb844b9c8b88b/pycrate_asn1c/asnobj.py#L5419

mitshell, 2018-05-14:
Fix with commit: https://github.com/ANSSI-FR/pycrate/commit/e34c088fb484cad6966b725e402073e2e04f76d6

closed 2018-05-14 by mitshell