P1sec/pycrate #27: Weird behavior when parsing a SCCP payload and trying to extract GT

p1-mmr, 2018-09-17:
Hello,

When parsing a SCCP payload:

```python
from pycrate_mobile.SCCP import parse_SCCP

sccp, _ = parse_SCCP(bytes.fromhex('0901030e190b12080012041491500050440b12080012042143650070055e625c4804000200006b1a2818060700118605010101a00d600ba1090607040000010019026c38a13602010102012e302e800589674523f18406a121436587f9041d24090421436587f90400819071516363000ce7f79b0c6abfe5eeb4fb0c'))
```

If we try to print the verbose representation of a Calling or Called Party Address structure, we can see that four "GT" objects are contained at the same level, three of which are called "GT [transparent]" (I guess that this is meant to correspond to the four existing types of GTs, of which one may be used at a time as a Calling or Called address):

```
>>> print(sccp['CalledPartyAddr'])
<CalledPartyAddr : <Len : 11><Value : <AddrInd : <res : 0><RoutingInd : 0 (route on GT)><GTInd : 4 (global title includes translation type, numbering plan, encoding scheme and nature of address indicator)><SSNInd : 1><PCInd : 0>><PC [transparent] : 0><SSN : 8 (mobile switching centre (MSC))><GT [transparent] : <OE : 0 (even number of address signals)><NAI : 1 (subscriber number)><Addr : >><GT [transparent] : <TranslationType : 0><Addr : 0x>><GT [transparent] : <TranslationType : 0><NumberingPlan : 1 (ISDN/telephony numbering plan (ITU-T E.163 and E.164))><EncodingScheme : 1><Addr : ><Addr [transparent] : 0x>><GT : <TranslationType : 0><NumberingPlan : 1 (ISDN/telephony numbering plan (ITU-T E.163 and E.164))><EncodingScheme : 2><spare : 0><NAI : 4 (international number)><Addr : 411905000544><Addr [transparent] : 0x>>>>
```

Only the used type is used when we print a prettified representation with `show()`:

```python
>>> print(sccp['CalledPartyAddr']['Value'].show())
### Value ###
 ### AddrInd ###
  <res : 0>
  <RoutingInd : 0 (route on GT)>
  <GTInd : 4 (global title includes translation type, numbering plan, encoding scheme and nature of address indicator)>
  <SSNInd : 1>
  <PCInd : 0>
 <SSN : 8 (mobile switching centre (MSC))>
 ### GT ###
  <TranslationType : 0>
  <NumberingPlan : 1 (ISDN/telephony numbering plan (ITU-T E.163 and E.164))>
  <EncodingScheme : 2>
  <spare : 0>
  <NAI : 4 (international number)>
  <Addr : 411905000544>
```

However, when trying to obtain string, we obtain an empty entry instead of the actual GT (here 411905000544):

```
>>> print(sccp['CalledPartyAddr']['Value']['GT'].show())
### GT [transparent] ###
 <OE : 0 (even number of address signals)>
 <NAI : 1 (subscriber number)>
 <Addr : >

>>> sccp['CalledPartyAddr']['Value']['GT']['Addr']
<Addr : >
```

Is this normal? Is there a way to retrieve the actual GT programatically?

Thank you

p1-bmu, 2018-09-18:
Yep, there are 4 types of GT defined for SCCP. When building or parsing a packet, 1 get instantiated, the 3 others are kept "transparent", depending of the GTInd (GT indicator) in the address header. Those 4 types have the same name "GT", hence calling it by its name does not guarantee that you get the one that is set.

A specific method was introduced in _SCCPAddr, CalledPartyAddr and CallingPartyAddr recently: .get_gt() which always return the GT that is set.

```python
>>> print(m['CalledPartyAddr'].get_gt().show())
### GT ###
 <TranslationType : 0>
 <NumberingPlan : 1 (ISDN/telephony numbering plan (ITU-T E.163 and E.164))>
 <EncodingScheme : 2>
 <spare : 0>
 <NAI : 4 (international number)>
 <Addr : 411905000544>

>>> print(m['CalledPartyAddr']['Value'].get_gt().show())
### GT ###
 <TranslationType : 0>
 <NumberingPlan : 1 (ISDN/telephony numbering plan (ITU-T E.163 and E.164))>
 <EncodingScheme : 2>
 <spare : 0>
 <NAI : 4 (international number)>
 <Addr : 411905000544>
```

p1-mmr, 2018-09-18:
I will use this, thank you.

closed 2018-09-18 by p1-bmu