P1sec/pycrate #41: RecursionError during decoding of Huawei cdr file

edsase, 2019-04-03:
I am trying to decode a Huawei cdr file using [this asn1 definition](https://drive.google.com/file/d/19dfbl6TUEZDNm61zEm8UEp3qNco8SoGH/view?usp=sharing) of it, which has been compiled into a python module named asn1_obj_pycrate using the command line pycrate_asn1compile.py. My code is as follows to decode ["my_cdr_file"](https://drive.google.com/file/d/19dfbl6TUEZDNm61zEm8UEp3qNco8SoGH/view?usp=sharing):
`
from asn1_obj_pycrate import CS_R99_R4_CHARGING
from binascii import hexlify, unhexlify
with open(my_cdr_file, 'rb') as fin:
    encoded_ber = hexlify(fin.read())
  ce_data_file.from_ber(encoded_ber)
` 
This code results in RecursionError: maximum recursion depth exceeded in comparison.
My goal is to extract the data in the cdr file into csv or xml.
Will appreciate some pointers on how to get this right.
Thanks.

edsase, 2019-04-04:
Hi, I was able to resolve the recursionError. It was caused by the way the file was opened. But now I have the issue that the returned records are in bytes string. Like  'callingNumber': b'\x912\x82\x98Y\x92\xf2'. Is there a way to have the returned values in utf8 or ascii text instead of bytes string? Thanks

p1-bmu, 2019-04-04:
In order to help you, I would need to reproduce your setup : can you send me the message or the file your are trying to process ? Without this, it will be hard to help.

edsase, 2019-04-04:
Thanks for your response.
[Here's the asn1 definition file](https://drive.google.com/file/d/19dfbl6TUEZDNm61zEm8UEp3qNco8SoGH/view?usp=sharing)
[Here's the file I want to decode.](https://drive.google.com/file/d/19dfbl6TUEZDNm61zEm8UEp3qNco8SoGH/view?usp=sharing)

p1-bmu, 2019-04-04:
Both links are pointing to the same .asn file.

edsase, 2019-04-04:
Sorry about that.
[Here's the file to be decoded.
](https://drive.google.com/file/d/1ibNekvIGXrmW7-bYENRBNMXkDwdNWDIJ/view?usp=sharing)
Thanks

p1-bmu, 2019-04-04:
Everything that is defined as OCTET STRING within the ASN.1 file will be decoded as such. This is the case for most of the data into your ASN.1 description file.
Regarding call numbers, they are defined as BCDDirectoryNumber (which is OCTET STRING) but with an explicit comment that says it is the same formatting as call number within TS 24.008 specification. For this you can use the following module from Pycrate:
```
>>> from pycrate_mobile.TS24008_IE import *
>>> # CalledPartyBCDNumber and CallingPartyBCDNumber are available from this module
>>> num = CallingPartyBCDNumber()
>>> num.from_bytes(b'\x912\x82\x98Y\x92\xf2')
>>> print(num.show())
### CallingPartyBCDNumber ###
 <Ext : 1>
 <Type : 1 (international number)>
 <NumberingPlan : 1 (ISDN / telephony numbering plan (E.164 / E.163))>
 <Num : 23288995292>
>>> num['Num'].decode()
'23288995292'
>>> # this is a string, do whatever you want with it
```
Hope this helps.

edsase, 2019-04-04:
This certainly helped! Thanks so much @p1-bmu

closed 2019-04-05 by p1-bmu

subodh7591, 2021-06-10:
> RecursionError: maximum recursion depth exceeded in comparison.

How did you solve this error??

subodh7591, 2021-06-10:
from config import TEMP_DIR
from Hardcore import CS_R99_R4_CHARGING as HardcoreSyntax
from binascii import hexlify, unhexlify
import sys
import time
print(sys.getrecursionlimit())

encoded = open(r"..\cdr_africell.dat","rb")
# encoded_ber = hexlify(encoded)
Final = HardcoreSyntax.CallEventDataFile 
print("here",Final.from_ber((encoded.read())))
This is my code. I am trying to decode the file but I get None . When I use hexlify it gives recusion limit exceeded error and if I don't use hexlify the output is None . Please help.

Alhamdou, 2023-11-24:
@p1-bmu am practically having the same error message but in my case is a bit different.
e
I need help with regards to decoding this file. In this code when I tried to decode the `MTCallRecord I did see an empty { } in my save.txt file.

This is my code:
import asn1tools

Compile ASN.1 schema
path_asn = 'cdr_schema.asn'
schema = asn1tools.compile_files(path_asn, codec='der')

Input file path
file_path = 'b06044079.dat'

Output file path
output_file_path = 'output_decode.txt'

with open(file_path, "rb") as f:
bit_cdr = f.read()

Decode the data
decoded_data = schema.decode('MTCallRecord', bit_cdr)

Save the decoded data to a text file
with open(output_file_path, "w") as output_file:
output_file.write(str(decoded_data))

print(f"Decoded data saved to: {output_file_path}")

but when I change the type that i want to decode let's say for instance to CallEventRecord i get an error like this. " raise MissingMandatoryFieldError(member, offset)
asn1tools.codecs.ber.MissingMandatoryFieldError: CallEventRecord.outGatewayRecord:"

If i also try to change the decoding to CallEventDataFIle i do get output but almost all the data in the output file will still be encoded.

I will definitely appreciate it if assisted.
link to the ASN file am using: https://drive.google.com/file/d/19dfbl6TUEZDNm61zEm8UEp3qNco8SoGH/view?usp=sharing
.dat link: https://drive.google.com/file/d/1ibNekvIGXrmW7-bYENRBNMXkDwdNWDIJ/view?usp=sharing

Thanks.

@Alhamdou

p1-bmu, 2023-11-24:
Thanks for your feedback.
This issue seems however related to https://github.com/eerimoq/asn1tools and not pycrate.