P1sec/pycrate #42: Compatibility with Python 3.6

DuongLe, 2019-04-05:
I could use the ASN.1 compiler fine with Python2.7 but faced following issues with Python3.6

In pycrate/tools/pycrate_asn1compile.py line 157: I had to change it from fd = open(f) to fd = open(f, 'rb') to read all the input files

I still got 
Traceback (most recent call last):
  File "pycrate/tools/pycrate_asn1compile.py", line 185, in <module>
    sys.exit(main())
  File "pycrate/tools/pycrate_asn1compile.py", line 173, in main
    compile_text(txt, **ckw)
  File "/home/ec2-user/anaconda3/lib/python3.6/site-packages/pycrate_asn1c/asnproc.py", line 209, in compile_text
    raise(ASN1Err('[proc] need only textual definition'))
pycrate_asn1c.err.ASN1Err: [proc] need only textual definition

p1-bmu, 2019-04-08:
The error thrown by the compiler corresponds to an issue in the file encoding, which is not recognized as a text file. On the other side, when you open a file with 'rb' (read-binary), it reads the whole file including null bytes, which is not compatible with a text file...
Please ensure that your ASN.1 files are properly encoded in UTF-8 or ascii before trying to compile them. Then submit again the error case (in case there is still an error) without changing the source code.
Thanks

DuongLe, 2019-04-08:
Thanks @p1-bmu ! After I made sure all of my ASN.1 files are encoded in UTF-8, no errors occurred while compiling

closed 2019-04-08 by DuongLe