P1sec/pycrate #46: "Raw" values output with JER encoding

demsjf8, 2019-05-06:
Hi,

I have not yet figured out to do this myself, so any suggestions would be appreciated:

Currently, the JER encoder "pretty prints" some basic values, e.g. "unavailable" in the following example from J2735:

```
   "brakes": {
    "abs": "unavailable",
    "auxBrakes": "unavailable",
    "brakeBoost": "unavailable",
    "scs": "unavailable",
    "traction": "unavailable",
    "wheelBrakes": "80"
   },
```

Is there a way to have the "raw" value (not sure that is the correct term) printed instead, like in the following?

```
         "brakes":{
            "wheelBrakes":"10000",
            "traction":0,
            "abs":0,
            "scs":0,
            "brakeBoost":0,
            "auxBrakes":0
         },
```

E.g. in the above example print the integer value of the ENUMs (0 instead of "unavailable") and the binary representation of the BIT STRING ("10000" instead of "80").

Thanks!

p1-bmu, 2019-05-06:
Hello,

the JER encoder outputs the values in the format specified by JER. You can change this, but it won't conform to JER anymore. JER encodings are defined in *_to_jval()* methods, for each ASN.1 object of the runtime (in *pycrate_asn1rt*).

It's here for ENUMERATED: https://github.com/P1sec/pycrate/blob/c28cb99741b2ee3ccbf90e959538f190a425d316/pycrate_asn1rt/asnobj_basic.py#L1228
And here for INTEGER: https://github.com/P1sec/pycrate/blob/c28cb99741b2ee3ccbf90e959538f190a425d316/pycrate_asn1rt/asnobj_basic.py#L524

You can hack those methods to fit your need.

demsjf8, 2019-05-07:
Hello,

Thanks for the quick reply, the clarification regarding JER, and the coding hints!

I will study JER spec again and consider whether maybe using actual JER-compliant encoding is more what I want than what I had in mind previously. That would also avoid the need to hack the standard-compliant code to get proprietary output.

Thanks again!

closed 2019-05-07 by demsjf8