P1sec/pycrate #48: Hexadecimal string data cannot be parsed

w910820618, 2019-06-21:
The hexadecimal data of the NGSetupRequest message I picked up with wireshark cannot be parsed.
`import ngap
import binascii

if __name__ == "__main__":
    content = "00150041000004001b00090002f89850000002010052000a03807465737420674e4200150001400066001a00000100010002f8980002100800006410100000641018000064000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
    # with open("ngsetup2.txt", "r") as f:
    #     content = f.read()
    # ngap.NGAP_PDU_Contents.NGSetupRequest.from_aper(bytes(content))
    bytecontent = bytes().fromhex(content)
    ngap.NGAP_PDU_Contents.NGSetupRequest.from_aper(bytecontent)`

ASN1 struct:
('SEQUENCE', {
protocolIEs: ('SEQUENCE OF', ('SEQUENCE', {
 id: 'INTEGER',
 criticality: 'ENUMERATED',
 value: ('OPEN_TYPE', {
  ('NGAP-IEs', 'GlobalRANNodeID'): ('CHOICE', {
   globalGNB-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    gNB-ID: ('CHOICE', {
     gNB-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   globalNgENB-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    ngENB-ID: ('CHOICE', {
     macroNgENB-ID: 'BIT STRING',
     shortMacroNgENB-ID: 'BIT STRING',
     longMacroNgENB-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   globalN3IWF-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    n3IWF-ID: ('CHOICE', {
     n3IWF-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   choice-Extensions: ('SEQUENCE', {
    id: 'INTEGER',
    criticality: 'ENUMERATED',
    value: 'OPEN_TYPE'
    })
   }),
  GlobalRANNodeID: ('CHOICE', {
   globalGNB-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    gNB-ID: ('CHOICE', {
     gNB-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   globalNgENB-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    ngENB-ID: ('CHOICE', {
     macroNgENB-ID: 'BIT STRING',
     shortMacroNgENB-ID: 'BIT STRING',
     longMacroNgENB-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   globalN3IWF-ID: ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    n3IWF-ID: ('CHOICE', {
     n3IWF-ID: 'BIT STRING',
     choice-Extensions: ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      value: 'OPEN_TYPE'
      })
     }),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    }),
   choice-Extensions: ('SEQUENCE', {
    id: 'INTEGER',
    criticality: 'ENUMERATED',
    value: 'OPEN_TYPE'
    })
   }),
  ('NGAP-IEs', 'RANNodeName'): 'PrintableString',
  RANNodeName: 'PrintableString',
  ('NGAP-IEs', 'SupportedTAList'): ('SEQUENCE OF', ('SEQUENCE', {
   tAC: 'OCTET STRING',
   broadcastPLMNList: ('SEQUENCE OF', ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    tAISliceSupportList: ('SEQUENCE OF', ('SEQUENCE', {
     s-NSSAI: ('SEQUENCE', {
      sST: 'OCTET STRING',
      sD: 'OCTET STRING',
      iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
       id: 'INTEGER',
       criticality: 'ENUMERATED',
       extensionValue: ('OPEN_TYPE', {})
       }))
      }),
     iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      extensionValue: ('OPEN_TYPE', {})
      }))
     })),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    })),
   iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
    id: 'INTEGER',
    criticality: 'ENUMERATED',
    extensionValue: ('OPEN_TYPE', {})
    }))
   })),
  SupportedTAList: ('SEQUENCE OF', ('SEQUENCE', {
   tAC: 'OCTET STRING',
   broadcastPLMNList: ('SEQUENCE OF', ('SEQUENCE', {
    pLMNIdentity: 'OCTET STRING',
    tAISliceSupportList: ('SEQUENCE OF', ('SEQUENCE', {
     s-NSSAI: ('SEQUENCE', {
      sST: 'OCTET STRING',
      sD: 'OCTET STRING',
      iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
       id: 'INTEGER',
       criticality: 'ENUMERATED',
       extensionValue: ('OPEN_TYPE', {})
       }))
      }),
     iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
      id: 'INTEGER',
      criticality: 'ENUMERATED',
      extensionValue: ('OPEN_TYPE', {})
      }))
     })),
    iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
     id: 'INTEGER',
     criticality: 'ENUMERATED',
     extensionValue: ('OPEN_TYPE', {})
     }))
    })),
   iE-Extensions: ('SEQUENCE OF', ('SEQUENCE', {
    id: 'INTEGER',
    criticality: 'ENUMERATED',
    extensionValue: ('OPEN_TYPE', {})
    }))
   })),
  ('NGAP-IEs', 'PagingDRX'): 'ENUMERATED',
  PagingDRX: 'ENUMERATED',
  ('NGAP-IEs', 'UERetentionInformation'): 'ENUMERATED',
  UERetentionInformation: 'ENUMERATED'
  })
 }))
})
[](url)

p1-bmu, 2019-06-24:
It seems you are not using the correct ASN.1 object to decode the NGAP message. Here is what I have:

```
In [1]: from pycrate_asn1dir import NGAP

In [2]: PDU = NGAP.NGAP_PDU_Descriptions.NGAP_PDU

In [3]: PDU.from_aper(unhexlify("00150041000004001b00090002f89850000002010052000a03807465737420674e4200150001400066001a00000100010002f89800021008000064101
   ...: 000006410180000640000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ...: 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"))

In [4]: print(PDU.to_asn1())
initiatingMessage : {
  procedureCode 21,
  criticality reject,
  value NGSetupRequest: {
    protocolIEs {
      {
        id 27,
        criticality reject,
        value GlobalRANNodeID: globalGNB-ID : {
          pLMNIdentity '02F898'H,
          gNB-ID gNB-ID : '00000201'H
        }
      },
      {
        id 82,
        criticality reject,
        value RANNodeName: "test gNB"
      },
      {
        id 21,
        criticality reject,
        value PagingDRX: v128
      },
      {
        id 102,
        criticality reject,
        value SupportedTAList: {
          {
            tAC '010001'H,
            broadcastPLMNList {
              {
                pLMNIdentity '02F898'H,
                tAISliceSupportList {
                  {
                    s-NSSAI {
                      sST '01'H,
                      sD '000064'H
                    }
                  },
                  {
                    s-NSSAI {
                      sST '02'H,
                      sD '000064'H
                    }
                  },
                  {
                    s-NSSAI {
                      sST '03'H,
                      sD '000064'H
                    }
                  }
                }
              }
            }
          }
        }
      }
    }
  }
}

In [5]: hexlify(PDU.to_aper())
Out[5]: b'00150041000004001b00090002f89850000002010052000a03807465737420674e4200150001400066001a00000100010002f8980002100800006410100000641018000064'
```

Looks OK to me. Take care also to provide only the NGAP message buffer, without padding bits (as it is the case with your hexadecimal buffer here).

w910820618, 2019-06-24:
Thank you, my method with you has solved my problem.

closed 2019-06-24 by w910820618