P1sec/pycrate #49: An error occurred while trying to import a NAS asn file

w910820618, 2019-07-04:
Hello, I want to ask you two questions.
1, I saw your library is to support 2G to 3G NAS message analysis, I would like to ask if the support does not support 5G NAS message parsing. I can't pass my NAS message in your program.
`python
import gNB.NGAP as NGAP
import binascii
from pycrate_mobile.NAS import *

PDU = NGAP.NGAP_PDU_Descriptions.NGAP_PDU

PDU.from_aper(binascii.unhexlify(
    "000f00814a000008005500020080002600810281007e02cd5c1994fe7e02410fd0cc1001032e0211225201000200000003170201024002000150020002b12b01022502000118010151010270000134817b00013491530222117100013400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007900121002f8290000007002f82901f82902f82906005a000100001a0007301bc0746d7369000300024040007000010000000005028001020300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002114000000000000b8718445ce7f0000b8718445ce7f00004054ed01000000004054ed01000000000080adfbce7f0000906fed0100000000906fed0100000000906fed0100000000906fed01000000003670ed0100000000908fed01000000000000000000000000908fed0100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007055ed010000000000000000000000000000000000000000ffffffffffffffff000000000000000000000000000000000000000000000000ffffffff0000000000000000000000000000000000000000e0538445ce7f000030615045ce7f0000f0645045ce7f0000781e297aff7f0000801e297aff7f00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010020000000000001002000000000000000008005500020080002600810281007e02cd5c1994fe7e02410fd0cc1001032e0211225201000200000003170201024002000150020002b11100000000000098718445ce7f000098718445ce7f0000b056ed0100000000b056ed01000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007900121002f8290000007002f82901f82902f82906005a000100001a0007301bc0746d7369000300024040007000010000000005028001020300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e10f000000000000786b8445ce7f00005036ed010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"))

print(PDU.to_asn1())
`
This is after my message is running, you can get the data in the NAS-PDU. I want to know if the data in the NAS-PDU is hexadecimal data? Or other coded data?

2、I tried to import the asn file I found on [this site](https://www.obj-sys.com/products/asn1apis/lte_3gpp_apis.php) about nas, but I can't import it into the program. Can you help me find out what went wrong?

p1-bmu, 2019-07-05:
1. Pycrate has currently no support for the 5G NAS (defined in TS 24.501). It only supports 2G-3G (mostly from TS 24.008) and 4G NAS messages (from TS 24.301).

2. NAS is not defined in ASN.1 syntax, so I guess that is the reason you cannot compile it with the pycrate ASN.1 compiler. I checked obj-sys web site, but access to their NAS document and SDK requires a registration. Sorry, but I won't investigate this unless you provide me those files directly.

w910820618, 2019-07-08:
Hello, I want to ask if the NAS library is written by auto-generated or self-encoded.

p1-bmu, 2019-07-08:
No, it's written by me, after reading the specifications !

w910820618, 2019-07-09:
I would like to ask if I use pycrate_asn1compile to parse the S1AP asn1 file. The error is pycrate_asn1c.err.ASN1ProcTextErr: TooEarlyInterRATHOReportReportFromEUTRAN.mobilityInformation: ASN1RefType to MobilityInformation, undefined. Can you tell me why?
My asn1 file is downloaded from wireshark's github repository at https://github.com/wireshark/wireshark/tree/master/epan/dissectors/asn1/s1ap.
I am looking forward to hearing from you.
Thank your!

w910820618, 2019-07-09:
I want to ask you what is the idea of writing a NAS library in the agreement? We need to implement the 5G NAS library now, can you help me with some ideas?

p1-bmu, 2019-07-09:
1) About S1AP: for any questions relating to something else than this issue's topic, please start a new issue.
2) About implementing a 5G NAS handler: I don't provide any specific support for such big features on my free time : contributions are always welcome, in case you want to do it by yourself (just do something similar to what I have done in pycrate_mobile/TS24301_*.py for LTE). And in case you just want some timely / commercial support, you can contact my employer P1 Security.

closed 2019-07-10 by w910820618