P1sec/pycrate #51: APER decoding Issue when 0 sized INTEGER

rwincey, 2019-07-17:
I have found inconsistencies when decoding APER encoded objects that contain Integers that have been encoded with lengths fields of 0. Microsoft's implementation seems to have the expectation that 0 means at least 1. The instance of this I'm seeing this in is in the Microsoft RDP protocol, https://winprotocoldoc.blob.core.windows.net/productionwindowsarchives/MS-RDPBCGR/%5bMS-RDPBCGR%5d.pdf, page 334 - Section 4.1.7. Server MCS Attach-User Confirm PDU, with the AttachUserConfirm::initiator field. When I parse this using APER, the initiator field returns None because ldet in the following code in codecs.py returns 0.

    def decode_intunconst(cla, char, offset=None):
        if cla.ALIGNED and cla._off[-1] % 8:
            cla.decode_pad(char)
        # 1) get byte-length determinant
        ldet = cla.decode_count(char)

I'm not sure if this is the expected behavior or a inconsistency in Microsoft's implementation of APER.  Either way I figured I'd bring it to your attention. For my purposes, I have added 

      #
      # Added because minimum byte size for an integer should be 1
      #
      if ldet == 0:
          ldet = 1

to decode_intunconst in codecs.py as a work-around to get my parser working.

p1-bmu, 2019-07-18:
Thanks for reporting. Could you provide an example of such encoding and decoding ?

p1-bmu, 2019-07-18:
I checked the ITU-T X.690 spec and pycrate's code: my understanding is that a simple null byte is not a valid PER encoding for an unconstrained integer. On the other side, it seems some tools (e.g. asn1-playground) accept to decode a null byte as the null value for such unconstrained integer.
Your patch proposal does not exactly that however: it requires 2 null bytes to decode to a null value.

Definitely, any examples here will help to decide how it patch the PER decoder in the best way !

rwincey, 2019-07-18:
The example I am working off of is from the reference document above. It's probably easier to look at the doc but I'll paste it in here. The APER encoding portion of interest is the AttachUserConfirm pdu. It's schema is defined at https://www.itu.int/rec/T-REC-T.125-199802-I/en.

4.1.7 Server MCS Attach-User Confirm PDU
The following is an annotated dump of the MCS Attach User Confirm PDU (section 2.2.1.7). 

00000000 03 00 00 0b 02 f0 80 2e 00 00 06 ........... 

03 00 00 0b -> TPKT Header (length = 11 bytes) 02 f0 80 -> X.224 

Data TPDU PER encoded (ALIGNED variant of BASIC-PER) PDU contents: 
2e 00 00 06 

0x2e: 
0 - --\
0 - |
1 - | CHOICE: From DomainMCSPDU select attachUserConfirm (11) 
0 - | of type AttachUserConfirm 
1 - | 
1 - --/ 
1 - AttachUserConfirm::initiator present 
0 - --\  | 
0x00:    | AttachUserConfirm::result = rt-successful (0) 
0 - | 
0 - | 
0 - --/ 
0 - padding 
0 - padding 
0 - padding 
0 - padding 
0 - padding 

0x00: 
0 - --\ 
0 - | 
0 - | 
0 - | 
0 - | 
0 - | 
0 - | 
0 - | 
      | AttachUserConfirm::initiator = 0x0006 + 1001 = 0x03ef = 1007 (user channel) 
0x06: | 
0 - | 
0 - | 
0 - | 
0 - | 
0 - | 
1 - | 
1 - | 
0 - --/

I created an asn1-playground schema below to help out. But as you can see above, a NULL byte proceeds the INTEGER value, AttachUserConfirm::initiator, and so the APER decoder in pycrate chokes and returns None for initiator.

World-Schema DEFINITIONS AUTOMATIC TAGS ::= 
BEGIN
  UserId ::= INTEGER -- created by Attach-User
  Result ::= ENUMERATED -- in Connect, response, confirm
  {
  rt-successful (0),
  rt-domain-merging (1),
  rt-domain-not-hierarchical (2),
  rt-no-such-channel (3),
  rt-no-such-domain (4),
  rt-no-such-user (5),
  rt-not-admitted (6),
  rt-other-user-id (7),
  rt-parameters-unacceptable (8),
  rt-token-not-available (9),
  rt-token-not-possessed (10),
  rt-too-many-channels (11),
  rt-too-many-tokens (12),
  rt-too-many-users (13),
  rt-unspecified-failure (14),
  rt-user-rejected (15)
}
  AttachUserConfirm ::= [APPLICATION 11] IMPLICIT 
  SEQUENCE
  {
  result Result,
  initiator UserId OPTIONAL
  }                                                   
END

p1-bmu, 2019-07-19:
I have imported ITU-T T124, 125 and 128 ASN.1 modules into pycrate_asn1dir. So, you can now simply:
```
In [1]: from pycrate_asn1dir.T125 import *

In [2]: p = MCS_PROTOCOL.DomainMCSPDU

In [3]: p.from_aper(b'\x2e\x00\x00\x06')

In [4]: print(p.to_asn1())
attachUserConfirm : {
  result rt-successful,
  initiator 1007
}

In [5]: hexlify(p.to_aper())
Out[5]: b'2e000006'
```

I also fixed the null length determinant issue inside the PER decoding of integers ; but I am not sure this was required to encode / decode the structure you indicated.

rwincey, 2019-07-19:
Checked out your commits and everything seems to work fine. I switched to the schemas you imported and everything seems to work with or without the null length fix you added, so maybe the issue originated from my manual conversions and imports of the schemas. Thanks for really going beyond the issue there. Saved me a couple imports :)

closed 2019-07-19 by rwincey