P1sec/pycrate #52: Ability to return bytes that are not consumed during a decoding operation

rwincey, 2019-07-19:
In certain circumstances, a byte stream with multiple variably encoded objects is encountered. Is there currently a way to return how many bytes were consumed from an operation like "from_aper"? This way the entire buffer can be passed to the decoder and it will decode the required number of bytes to decode the first object and then pass back the remainder. My current workaround has "from_aper" in  "asnobj.py" return the remainder of bytes and "_from_per" in "asnobj_construct.py" return "char.to_bytes()".  Is there better way to do this?

p1-bmu, 2019-07-23:
You can pass a Charpy instance directly to from_aper() / from_uper(), instead of a bytes buffer:
https://github.com/P1sec/pycrate/blob/9f94f7a2d9c9bf90d875f9811d859924b0d375aa/pycrate_asn1rt/asnobj.py#L1182

It will be consumed during the parsing of the ASN.1 object ; you will be able to pass it to several successive objects then. How to use Charpy() is described here: https://github.com/P1sec/pycrate/wiki/Using-pycrate-core-objects#charpy

rwincey, 2019-07-23:
Awesome, thank you.

closed 2019-07-23 by rwincey