P1sec/pycrate #53: Improve 5G NAS messaging

w910820618, 2019-07-24:
I want to improve the 5G NAS message, I hope you can give me some tips. For example, what logic did you implement for 4G NAS messages before?

w910820618, 2019-07-25:
```python
class EMMAttachAccept(Layer3):
    _GEN = (
        EMMHeader(val={'Type':66}),
        Uint('spare', bl=4),
        Type1V('EPSAttachResult', dic=EPSAttRes_dict),
        Type3V('T3412', val={'V':b'\0'}, bl={'V':8}, IE=GPRSTimer()),
        Type4LV('TAIList', val={'V':6*b'\0'}, IE=TAIList()),
        Type6LVE('ESMContainer', val={'V':b'\0\0\0'}),
        Type4TLV('GUTI', val={'T':0x50, 'V':b'\xf6'+10*b'\0'}, IE=EPSID()),
        Type3TV('LAI', val={'T':0x13, 'V':5*b'\0'}, bl={'V':40}, IE=LAI()),
        Type4TLV('ID', val={'T':0x23, 'V':b'\xf4\0\0\0\0'}, IE=ID()),
        Type3TV('EMMCause', val={'T':0x53, 'V':b'\0'}, bl={'V':8}, IE=EMMCause()),
        Type3TV('T3402', val={'T':0x17, 'V':b'\0'}, bl={'V':8}, IE=GPRSTimer()),
        Type3TV('T3423', val={'T':0x59, 'V':b'\0'}, bl={'V':8}, IE=GPRSTimer()),
        Type4TLV('EquivPLMNList', val={'T':0x4A, 'V':3*b'\0'}, IE=PLMNList()),
        Type4TLV('EmergNumList', val={'T':0x34, 'V':b'\x02\x01\0'}, IE=EmergNumList()),
        Type4TLV('EPSNetFeat', val={'T':0x64, 'V':b'\0\0'}, IE=EPSNetFeat()),
        Type1TV('AddUpdateRes', val={'T':0xF, 'V':0}, IE=AddUpdateRes()),
        Type4TLV('T3412Ext', val={'T':0x5E, 'V':b'\0'}, IE=GPRSTimer3()),
        Type4TLV('T3324', val={'T':0x6A, 'V':b'\0'}, IE=GPRSTimer()),
        Type4TLV('ExtDRXParam', val={'T':0x6E, 'V':b'\0'}, IE=ExtDRXParam()),
        Type1TV('SMSServStat', val={'T':0xE, 'V':0}, IE=SMSServStat())
        )
```
I compared the TS24.301 protocol, but I still don't know how you copied it. Can you explain? Can I add a social account if I can?

w910820618, 2019-07-25:
I want to know ( EMMHeader(val={'Type':66}),
         Uint('spare', bl=4)), val and bl are all assigned according to what protocol?

p1-bmu, 2019-07-25:
1) start to learn using pycrate:
https://github.com/P1sec/pycrate/wiki/Using-pycrate-core-objects
https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages

2) read the code in:
https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/TS24301_EMM.py
https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/TS24301_ESM.py
https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/TS24301_IE.py
Together with the 3GPP specification TS 24.301.

From here, you will be able to understand how each message class is built. This enables to encode and decode all LTE NAS messages.

3) do the same with TS 24.501.
This is for sure not an easy task ! But doable as soon as you take the time to understand how the library is working, and you get more familiar with its source code.

w910820618, 2019-07-25:
[image]
Can you tell me where I circled, are you assigning them according to the content of 3GPP? This is very helpful for me to understand the program, thank you very much.

w910820618, 2019-07-25:
How does parse_NAS_MO() handle the acquisition of NAS data? Let's talk about logic, not too much detail.

w910820618, 2019-07-25:
The two questions asked above are more critical to me, just give a hint. Can I understand the principle of pycrate work more quickly? Also ask why you are not using asn1 files to generate .py, does pycrate_asn1compile.py not support converting asn1 files to .py?

p1-bmu, 2019-07-26:
Sorry, however as I already answered in issue#49, I am not able to provide free support for such complex / time-consuming questions. In case you want professional support or specific development on pycrate, feel free to contact my employer P1 Security.

w910820618, 2019-07-26:
I have already contacted your employer, thank you for your help.

closed 2019-07-27 by w910820618