P1sec/pycrate #55: parsing NAS

w910820618, 2019-07-29:
Which method is the method of parsing NAS?

w910820618, 2019-07-29:
I ask, is your program working as I described it?
1. First determine which NAS signaling belongs to by parsing the NAS packet.
2. Load NAS signaling into memory
3. Use the _from_char method of each IE in the signaling to parse

p1-bmu, 2019-07-29:
Parsing NAS messages is explained in the wiki: https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages.
How it is done technically is a little bit complex, all NAS messages are subclass of Layer3, see https://github.com/P1sec/pycrate/blob/786a69b36f313c6ccfdf989bf5787e3e05e2c3a8/pycrate_mobile/TS24007.py#L68.

w910820618, 2019-07-30:
[image]
Why can't I assign an EPSID like this?

p1-scg, 2019-07-30:
p1-bmu 不可以随便帮助你

w910820618, 2019-07-30:
哦，知道了

closed 2019-07-30 by w910820618