P1sec/pycrate #59: NGAP (3GPP -38413) - Failure while decoding APER encoded hex string

UmakantKulkarni, 2019-08-19:
I've ngap data that is being encoded by another c++ asn compiler OSS (Nokalava) - 0000040082000a0c1dcd6500301dcd6500008b000a01f00a0a0b0b0800000200860001000088000700010000052d00

But I'm getting following error while decoding it:
`
import NGAP
from binascii import hexlify, unhexlify
PDU = NGAP.NGAP_PDU_Descriptions.NGAP_PDU
p = "0000040082000a0c1dcd6500301dcd6500008b000a01f00a0a0b0b0800000200860001000088000700010000052d00"
PDU.from_aper(unhexlify(p))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj.py", line 1190, in from_aper
    self._from_per(char)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 247, in _from_per
    Cho._from_per(char)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 795, in _from_per
    Comp._from_per(char)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj_ext.py", line 309, in _from_per
    val = ASN1CodecPER.decode_unconst_open(char, wrapped=Obj)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/codecs.py", line 864, in decode_unconst_open
    wrapped.from_aper(char)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj.py", line 1190, in from_aper
    self._from_per(char)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 767, in _from_per
    E = char.get_uint(1)
  File "/root/Umakant/n1n2Decoder/compiler/pycrate-master/pycrate_core/charpy.py", line 776, in get_uint
    .format(bitlen, self._len_bit-self._cur)))
pycrate_core.charpy.CharpyErr: bitlen overflow: 1, max 0
`
Any idea what's wrong here?
Compiled Python Library - [https://raw.githubusercontent.com/P1sec/pycrate/master/pycrate_asn1dir/NGAP.py](url)

UmakantKulkarni, 2019-08-20:
Any thoughts on this one?

p1-bmu, 2019-09-02:
The current pycrate NGAP ASN.1 model has been extracted from the 38.413 Rel.15.3.0 spec (see https://portal.3gpp.org/desktopmodules/Specifications/SpecificationDetails.aspx?specificationId=3223 for all NGAP releases). Beware that the different releases of NGAP are not compatible with each others, yet.

Which NGAP release did you used with your Nokalava compiler to generate this message ? To which value / structure does it correspond ?

UmakantKulkarni, 2019-09-03:
I compiled 38.413 Rel.15.2.0 - http://www.3gpp.org/ftp//Specs/archive/38_series/38.413/38413-f20.zip using tool pycrate_asn1compile. I've Nokalava compiled version of the same spec release.

This is what partially decoded message looks like:
`
             NGAP-PDU:
00 0.......   Extension of NGAP-PDU = 0 :Absent
   .00.....   NGAP-PDU = 0 :initiatingMessage
               InitiatingMessage:
                NGAP-ELEMENTARY-PROCEDURES:ProcedureCode:
   ...00000      Padding = 00000b
00 00000000      ProcedureCode = 0 :id-AMFConfigurationUpdate
04 00......     Criticality = 0 :reject
   ..000100     Padding = 000100b
00 0.......     Range of value length(octet) = 0 :Up to 127
   .0000000     value length(octet) = 0
`

mitshell, 2019-09-03:
Could you provide the complete value that you have encoded ? If possible in a readable format, e.g. asn1 or xer or gser...

On my side, I can see pycrate starting decoding a *AMFConfigurationUpdate*. However, the 1st IE has id 10, what corresponds to IE *AMF-UE-NGAP-ID* which is not part of the list of IEs for *AMFConfigurationUpdate*, and the 2nd IE has id 7, what corresponds to IE *AMF-TNLAssociationToRemoveList*. Pycrate fails during the decoding of this 2nd IE, saying some bits are missing from the provided buffer.

UmakantKulkarni, 2019-09-03:
Sure, let me check..

UmakantKulkarni, 2019-09-04:
This is what completely decoded data looks like:
`
value PDUSessionResourceSetupRequestTransfer ::= 
{
  protocolIEs 
  {
    {
      id 130,
      criticality reject,
      value PDUSessionAggregateMaximumBitRate : 
      {
        pDUSessionAggregateMaximumBitRateDL 500000000,
        pDUSessionAggregateMaximumBitRateUL 500000000
      }
    },
    {
      id 139,
      criticality reject,
      value UPTransportLayerInformation : gTPTunnel : 
      {
        transportLayerAddress '00001010 00001010 00001011 00001011'B,
        gTP-TEID '08000002'H
      }
    },
    {
      id 134,
      criticality reject,
      value PDUSessionType : ipv4
    },
    {
      id 136,
      criticality reject,
      value QosFlowSetupRequestList : 
      {
        {
          qosFlowIdentifier 1,
          qosFlowLevelQosParameters 
          {
            qosCharacteristics nonDynamic5QI : 
            {
              fiveQI 5
            },
            allocationAndRetentionPriority 
            {
              priorityLevelARP 12,
              pre-emptionCapability may-trigger-pre-emption,
              pre-emptionVulnerability not-pre-emptable
            }
          }
        }
      }
    }
  }
}
`

p1-bmu, 2019-09-05:
This works fine in pycrate:
```
In [1]: from pycrate_asn1dir.NGAP import *

In [2]: c = NGAP_IEs.PDUSessionResourceSetupRequestTransfer

In [3]: c.from_aper(unhexlify('0000040082000a0c1dcd6500301dcd6500008b000a01f00a0a0b0b0800000200860001000088000700010000052d00'))

In [4]: print(c.to_asn1())
{
  protocolIEs {
    {
      id 130,
      criticality reject,
      value PDUSessionAggregateMaximumBitRate: {
        pDUSessionAggregateMaximumBitRateDL 500000000,
        pDUSessionAggregateMaximumBitRateUL 500000000
      }
    },
    {
      id 139,
      criticality reject,
      value UPTransportLayerInformation: gTPTunnel : {
        transportLayerAddress '0A0A0B0B'H -- 
        
        

         --,
        gTP-TEID '08000002'H
      }
    },
    {
      id 134,
      criticality reject,
      value PDUSessionType: ipv4
    },
    {
      id 136,
      criticality reject,
      value QosFlowSetupRequestList: {
        {
          qosFlowIdentifier 1,
          qosFlowLevelQosParameters {
            qosCharacteristics nonDynamic5QI : {
              fiveQI 5
            },
            allocationAndRetentionPriority {
              priorityLevelARP 12,
              pre-emptionCapability may-trigger-pre-emption,
              pre-emptionVulnerability not-pre-emptable
            }
          }
        }
      }
    }
  }
}
```

On the other side, *PDUSessionResourceSetupRequestTransfer* is not a NGAP PDU (defined in NGAP-PDU-Descriptions.NGAP-PDU) but just an Information Element...

closed 2019-09-05 by p1-bmu

UmakantKulkarni, 2019-09-05:
Oh I see.
Thanks a lot.!

UmakantKulkarni, 2019-09-27:
Hi @p1-bmu , Quick question:
Any idea, how to encode this above message back to binary data?

p1-bmu, 2019-10-01:
You should read the wiki: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime

UmakantKulkarni, 2019-10-02:
Thanks, I tried one of the example, but now getting this error:
`
>>>
>>> decodedN2Data
{'protocolIEs': [{'criticality': 'reject', 'id': 130, 'value': {'pDUSessionAggregateMaximumBitRateDL': 500000000, 'pDUSessionAggregateMaximumBitRateUL': 500000000}}, {'criticality': 'reject', 'id': 139, 'value': {'gTPTunnel': {'gTP-TEID': '08000002', 'transportLayerAddress': {'length': 32, 'value': '0a0a0b0b'}}}}, {'criticality': 'reject', 'id': 134, 'value': 'ipv4'}, {'criticality': 'reject', 'id': 136, 'value': [{'qosFlowIdentifier': 1, 'qosFlowLevelQosParameters': {'allocationAndRetentionPriority': {'pre-emptionCapability': 'may-trigger-pre-emption', 'pre-emptionVulnerability': 'not-pre-emptable', 'priorityLevelARP': 12}, 'qosCharacteristics': {'nonDynamic5QI': {'fiveQI': 5}}}}]}]}
>>>
>>> ngapIes = NgapDec2018.NGAP_IEs.PDUSessionResourceSetupRequestTransfer
>>>
>>> ngapIes.set_val(decodedN2Data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/pycrate_asn1rt/asnobj.py", line 1097, in set_val
    self._safechk_val(self._val)
  File "/usr/local/lib/python3.6/site-packages/pycrate_asn1rt/asnobj_construct.py", line 591, in _safechk_val
    self._cont[k]._safechk_val(val[k])
  File "/usr/local/lib/python3.6/site-packages/pycrate_asn1rt/asnobj_construct.py", line 1998, in _safechk_val
    self._cont._safechk_val(v)
  File "/usr/local/lib/python3.6/site-packages/pycrate_asn1rt/asnobj_construct.py", line 591, in _safechk_val
    self._cont[k]._safechk_val(val[k])
  File "/usr/local/lib/python3.6/site-packages/pycrate_asn1rt/asnobj_ext.py", line 174, in _safechk_val
    raise(ASN1ObjErr('{0}: invalid value, {1!r}'.format(self.fullname(), val)))
pycrate_asn1rt.err.ASN1ObjErr: PDUSessionResourceSetupRequestTransfer.protocolIEs._item_.value: invalid value, {'pDUSessionAggregateMaximumBitRateDL': 500000000, 'pDUSessionAggregateMaximumBitRateUL': 500000000}
`
Any idea, what am I missing?

p1-bmu, 2019-10-03:
As said by the error message, you are not assigning the correct value to the PDUSessionAggregateMaximumBitRate_ IE:
{'pDUSessionAggregateMaximumBitRateDL': 500000000, 'pDUSessionAggregateMaximumBitRateUL': 500000000}
instead of
('PDUSessionAggregateMaximumBitRate', {'pDUSessionAggregateMaximumBitRateDL': 500000000, 'pDUSessionAggregateMaximumBitRateUL': 500000000})

Please verify what you do before submitting questions, and in case you face a real issue, please open a proper github issue instead of commenting on a closed one.