P1sec/pycrate #60: OER/COER support

JNevrly, 2019-08-28:
Obviously this is not an issue, but is there a plan to support OER/COER in a near future?

p1-bmu, 2019-09-02:
Currently, there is no plan to introduce a support of OER in pycrate. As this represent a large amount of work, and I have no need for it, I won't do it in my free time / in a near future.
In case you absolutely need it and won't implement it yourself, you can contact my employer (https://www.p1sec.com/corp/) to request a commercial support for such development.

p1-bmu, 2019-11-04:
Switching this as enhancement, but not expecting to work on it anytime soon.

JNevrly, 2020-09-10:
So I've progressed with the [OER/COER implementation](https://github.com/P1sec/pycrate/compare/master...JNevrly:OER_implementation), most of the common types are done, but I'm bit confused about the implementation of the TIME type. Not knowing much about ASN.1, this seems to be some new-ish type (since it's not mentioned in the old Dubuisson copy I have) and I don't see any support for it in the Pycrate. Is that observation correct? (I the OER/COER standard X.696 (08/2015), a significant portion is spent on the TIME type, so it looks like a big deal ;)).
To simplify the question - is it OK to skip the TIME type implementation and go ahead with the PR?

p1-bmu, 2020-09-10:
TIME was newly introduced in 2015 IIRC (together with OID-IRI and RELATIVE-OID-IRI), and seems quite complex also in BER/CER/DER. You can skip it as there is currently no support for it in pycrate, neither in the compiler nor in the runtime.

JNevrly, 2020-09-21:
With PR  #110 merged, I think this can be closed.

closed 2020-09-21 by JNevrly