P1sec/pycrate #61: Could not decode the X2AP message

jufei, 2019-10-14:
hi:
  I tried to use pycrate to decode a X2AP message, like below:

from pycrate_asn1dir import X2AP
PDU = X2AP.X2AP_PDU_Descriptions.X2AP_PDU
s = '201b008203000004006f0002000800cf0002000d00d240360000d540310a003b81f00a6ce901001a001303e00a6ce901401b0017007c0a6ce901601b001300071f000001012e400140013740014000d30081b581b3186c0ce0440ccae2078e3190437630d061e6407c0204a83c430106b7509cc613349a9e0026933a800000620c65d26a0c0a500a201b4064d202680001886ae32e932a31000d6267bfe2e0049de15bc2c5876f0f1625bc4c58d6f1b15414c09a70300442a03a0093ffea1db600203805930165004048204004808009302071a232c08966113704030000080801a500a201b40646d40007fffc000000043b8621800004041b84218000100b2802024102002404004981038d12b88960709242c1c1821c200e065c089665462001ac4cf7fc5c0093bc2b7858b0ede1e2c4b7898b1ade362a8298134e35524940408221061028004a01a01198068086601803132f68104cbd80b132f6c8230308807235600007f4210840080402581c2490b070608708038940e000010af7a008001000007a00098801c0080c0600203115514010800c12ca007e4026b143810086200072084d40005c0c0451000390426a0002e0802188001c82135008170501144000e4109a804082000000a1860406080b0382b001f9628000000200400182e001809006600b400016488822cf1fffff83f0340050a06946fd7e39b8e614000000002104d4f000'

PDU.from_aper(unhexlify(s))
print(PDU.to_asn1())

But I got the following output:

OPEN._from_per: X2AP-PDU.successfulOutcome.value, unable to retrieve a table-looked up object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
successfulOutcome : {
  procedureCode 27,
  criticality reject,
  value '000004006F0002000800CF0002000D00D240360000D540310A003B81F00A6CE901001A001303E00A6CE901401B0017007C0A6CE901601B001300071F000001012E400140013740014000D30081B581B3186C0CE0440CCAE2078E3190437630D061E6407C0204A83C430106B7509CC613349A9E0026933A800000620C65D26A0C0A500A201B4064D202680001886AE32E932A31000D6267BFE2E0049DE15BC2C5876F0F1625BC4C58D6F1B15414C09A70300442A03A0093FFEA1DB600203805930165004048204004808009302071A232C08966113704030000080801A500A201B40646D40007FFFC000000043B8621800004041B84218000100B2802024102002404004981038D12B88960709242C1C1821C200E065C089665462001AC4CF7FC5C0093BC2B7858B0EDE1E2C4B7898B1ADE362A8298134E35524940408221061028004A01A01198068086601803132F68104CBD80B132F6C8230308807235600007F4210840080402581C2490B070608708038940E000010AF7A008001000007A00098801C0080C0600203115514010800C12CA007E4026B143810086200072084D40005C0C0451000390426A0002E0802188001C82135008170501144000E4109A804082000000A1860406080B0382B001F9628000000200400182E001809006600B400016488822CF1FFFFF83F0340050A06946FD7E39B8E614000000002104D4F000'H
}

It seems could not decode it correctly.

Thanks.

p1-bmu, 2019-10-14:
The 1st part of your message corresponds to an X2AP procedure code 27, which is not defined even in the very last 3GPP release, and an undefined criticality. So I suppose the buffer you pass to the X2AP decode is not correct.
Can you please provide the corresponding / expected value corresponding to this message ?

jufei, 2019-10-14:
Hi:
                Thanks for your quickly response.

                See the attached pcap file:

                I used the packet which index = 46, SgNBAdditionRequestAcknowledge

                Wireshark can decode it：

[cid:image001.png@01D582AE.188A87F0]

jufei, 2019-10-14:
[image]
[image]

because I could not upload pcap file directoly, I renamed it to pcapfile.jpg

p1-bmu, 2019-10-14:
Can you please attach the pcap (and any other attachment) properly within this issue ? It seems your email attachments have been dropped.

jufei, 2019-10-14:
I have reattached it

p1-bmu, 2019-10-14:
OK, I misread the X2AP spec: this message with procedure code 27 is part of the 3GPP release 15.  Current pycrate support is mainly with release 13, so I need to update EUTRAN ASN.1 files : I'll do it soon.

jufei, 2019-10-14:
BTW, the document has a error:

Doc: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime

[cid:image001.png@01D582B0.22AC9B90]

The PDU is not defined, it should be:

PDU = S1AP.S1AP_PDU_Descriptions.S1AP_PDU
PDU.from_aper(xxxxx)

jufei, 2019-10-14:
Thanks a lot.

p1-bmu, 2019-10-14:
I just added support for eUTRAN rel.15. Now this should work properly.
Question from mine: do you mind if I use some of your pcap file parts to introduce new test suites ?

jufei, 2019-10-15:
Sure, you can use it.

Thanks a lot. I will install the whole lib again.

closed 2019-10-16 by p1-bmu