P1sec/pycrate #62: Could not decode the detail part of S1AP InitialUEMessage protocol IE Item2

jufei, 2019-10-15:
hi, 
   I have a pcap file that contains some s1ap message. For the first InitialUEMessage, pycrate could not decode the detail for it.

Here is the output of item 1:
      {
        id 26,
        criticality reject,
        value NAS-PDU: '170F0394AD060741010BF605F520C35101C0699AAE07F0700000100010002A020CD011D127238080211001000010810600000000830600000000000D00000A000005000010000011005205F52007085C0A005D0106E0C16F04F0007000'H
      },

The wireshark can decoded it as: in the pic
[image]
[image]

Thanks.

p1-bmu, 2019-10-15:
The NAS protocol is not part of S1AP. It is specified in TS 24.301 (for LTE) and is not using ASN.1. Pycrate provides a complete support for 2G-3G-4G NAS protocols encoding and decoding, based on the 3GPP release 12 or 13 (if I remember well), see:
https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages

So you need to call a proper NAS decoder to decode NAS-PDU within S1AP messages.

jufei, 2019-10-15:
Hi:
                Thanks for your quickly response.

                Yes , I see the attach request message definition in https://github.com/p1sec/pycrate/blob/master/pycrate_mobile/TS24301_EMM.py

                I tried to decode the first packet: attach request like below:

[root@oam_131 2]# python
Python 2.7.5 (default, Jul 13 2018, 13:06:57)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-28)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> s='000c40808600000500080002000d001a005e5d170f0394ad060741010bf605f520c35101c0699aae07f0700000100010002a020cd011d127238080211001000010810600000000830600000000000d00000a000005000010000011005205f52007085c0a005d0106e0c16f04f0007000004300060005f5200708006440080005f520000ef3200086400130'
>>> from pycrate_mobile.NAS import *
>>> Msg, err = parse_NAS_MO(unhexlify(s))
>>> err
97
>>> Msg, err = parse_NAS_MT(unhexlify(s))
>>> err
97
>>>

All failed.

Thanks.

https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages

p1-bmu, 2019-10-15:
You are not passing the proper buffer to the NAS decoder. Look at your wireshark screen capture: the NAS PDU starts with 0x170f03.

jufei, 2019-10-16:
OK and thanks, I will try again

jufei, 2019-10-16:
hi:
   Yes I can decode the NAS PDU with parse_NAS_MO, thanks.
   I am writing some test script to check the message from a pcap file. Is it possible to decode the packet like wireshark?

Thanks again.

mitshell, 2019-10-16:
This is up to you to develop your own application to mimic wireshark. Pycrate is just a library, use it in your own way.

closed 2019-10-16 by p1-bmu