P1sec/pycrate #64: to_json bug when converting from decimel to hex

junyian, 2019-10-24:
This is the code I'm testing with

sch = RRCLTE.EUTRA_RRC_Definitions.BCCH_DL_SCH_Message
sch.from_uper_ws(unhexlify('405454059C4A816A115818211081044C23C1001C002D'))

with sch(), the cellIdentity shows correctly as 135700757
with sch.to_asn1(), this converts correctly to 816A115
but with sch.to_json(), the hex becomes 816A1150

p1-bmu, 2019-10-25:
Thanks for the feedback.
JSON Encoding Rules is defined in ITU-T X.697. In section 24.2.1, it says the following about fixed size bit string encoding: "The encoding of a bitstring value with a fixed size shall be a JSON string. The Unicode character string denoted by the JSON string shall consist of an even number of the hexadecimal digits 0123456789abcdefABCDEF, with each consecutive pair of digits encoding one group of eight consecutive bits in the bitstring value."
Therefore, I believe the current JER encoding in Pycrate is correct.

junyian, 2019-10-27:
Hi there. I get it if the conversion becomes "0816A115", where the extra 0 becomes the prefix to meet the encoding rules and this doesn't change the value itself. But putting the "0" at the end (hence "816A1150") gives a totally different meaning to the hex number.

p1-bmu, 2019-10-28:
I understand your frustration, however what you propose would not be compliant with the ITU-T specification.
Personally, I am not willing to diverge from the JER specification for the json API within the ASN.1 runtime. In case you decide to implement your own JSON-based encoding-decoding rules, feel free to submit a PR.

As an alternative, you can also choose to encode the ASN.1 object value to json with the Python json encoder like this:
```
In [6]: import json

In [7]: json.dumps(sch())
Out[7]: '{"message": ["c1", ["systemInformationBlockType1", {"cellSelectionInfo": {"q-RxLevMin": -64}, "freqBandIndicator": 3, "schedulingInfoList": [{"sib-MappingInfo": ["sibType3"], "si-Periodicity": "rf16"}, {"sib-MappingInfo": ["sibType5"], "si-Periodicity": "rf32"}, {"sib-MappingInfo": ["sibType6"], "si-Periodicity": "rf64"}], "systemInfoValueTag": 1, "cellAccessRelatedInfo": {"cellBarred": "notBarred", "plmn-IdentityList": [{"plmn-Identity": {"mcc": [5, 1, 5], "mnc": [0, 2]}, "cellReservedForOperatorUse": "notReserved"}], "cellIdentity": [135700757, 28], "csg-Indication": false, "trackingAreaCode": [40010, 16], "intraFreqReselection": "allowed"}, "si-WindowLength": "ms40"}]]}'
```
But I suppose you will not be happy with the format of the CellID again.

junyian, 2019-10-28:
Actually I'm not frustrated. I just find it bizarre that the ITU-T specification would allow conversions by adding the 0 at the end.
The alternative you suggested looks interesting. It's not hex, that's true, but that's just a small matter as long as the underlying value is accurate. But the bitstring size there do throws off the format of the data.
Anyway, I think I've gotten what I need. So thanks for the responses. I shall be closing this item now.

closed 2019-10-28 by junyian