P1sec/pycrate #66: ASN.1 error with bit string and no option selected

KenWhitesell, 2019-10-24:
Sample asn.1 file (greatly reduced from J2735):

```
DSRC DEFINITIONS AUTOMATIC TAGS::= BEGIN 

Lane ::= SEQUENCE {
   directionalUse  LaneDirection
   }

LaneDirection ::= BIT STRING {
   ingressPath     (0), 
   egressPath      (1)
   }

END
```

This asn text compiles and works fine:
`{ directionalUse {ingressPath} }`

This text 
`{ directionalUse {} }`
generates this error:
```
Traceback (most recent call last):
  File "/opt/data/tskww/.ve/py38/lib/python3.8/site-packages/pycrate-0.4-py3.8.egg/pycrate_asn1rt/asnobj_construct.py", line 1205, in _from_asn1
    t_ident, t_rest = txt.split(' ', 1)
ValueError: not enough values to unpack (expected 2, got 1)
```

p1-bmu, 2019-10-25:
Could you please provide the complete textual definition you try to compile ? So that I can reproduce exactly your issue.

KenWhitesell, 2019-10-26:
I've attached a file which is a terminal session walking through the complete set of steps to recreate the problem identified here. Please let me know if you have any questions or need any further information.

[Xterm.log.tskww-HP-2.txt](https://github.com/P1sec/pycrate/files/3774262/Xterm.log.tskww-HP-2.txt)

p1-bmu, 2019-10-28:
OK, thanks.
This is related to passing ASN.1 BIT STRING values as named offsets. Here, the bug is that an empty set of offsets is not accepted by the runtime. A patch should be done somewhere here:
https://github.com/P1sec/pycrate/blob/5e7406d99346c2130b0e01f23c7c39a4756e857b/pycrate_asn1rt/asnobj_str.py#L223
When setting the BIT STRING value from the named offsets, we should also take care to set a correct bit length according to any potential SIZE constraint.

I will work on it when I have some time, maybe next week or so.

p1-bmu, 2019-11-04:
This commit should solve the issue: https://github.com/P1sec/pycrate/commit/afc9849c4c895ef8924c7c9dc77d0274bc6e735b

KenWhitesell, 2019-11-04:
**_Fantastic_** This commit addresses both issues I reported, and now allows me to encode/decode UPER-encoded J2735 TIM MessageFrames. (At least my three test cases.)
**Thank you!**

closed 2019-11-04 by KenWhitesell