P1sec/pycrate #67: UE can not attach to corenet

LucasCSTI, 2019-11-07:
eNB can attach to corenet but UE can not attach to corenet.

[image]

p1-bmu, 2019-11-07:
How am I supposed to debug your issue with such a picture ?
Please provide extended debugging information, e.g. a PCAP of the S1 interface, the complete logs provided by corenet (after setting everything in verbose mode: ENBd.TRACE_ASN_S1AP, UEd.TRACE_S1AP, UEd.TRACE_NAS_EPS_SEC, UEd.TRACE_NAS_EPS) ...

LucasCSTI, 2019-11-07:
I can only capture the pcpa file from S1 interface, and did not found the complete logs provided by corenet. Thanks.

[corenet_S1AP_20191107.zip](https://github.com/P1sec/pycrate/files/3818801/corenet_S1AP_20191107.zip)

p1-bmu, 2019-11-07:
OK, can you provide the complete Python stack trace of the error: no picture, just the text, but the complete one, not just the last part of it.

In order to configure the logging level of corenet, you can set the following variables to True in corenet.py:
ENBd.TRACE_ASN_S1AP = True
UEd.TRACE_S1AP = True
UEd.TRACE_NAS_EPS_SEC = True
UEd.TRACE_NAS_EPS = True
All the logs are written into the /tmp/corenet.log.

LucasCSTI, 2019-11-13:
Sorry for late responds, attached please find a copy of log file. Thanks.

[corenet.log](https://github.com/P1sec/pycrate/files/3838931/corenet.log)

p1-bmu, 2019-11-13:
Could you provide the complete stack trace of the bug you are encountering ? Without this, I cannot solve your issue.
Moreover, from the logs, it seems you did not configure your AuC database file as indicated in the README: "[2019-11-07 17:40:52.201] [WNG] [AuC] [make_4g_vector] IMSI 466666100000041 not present in AuC.db"

LucasCSTI, 2019-11-14:
Attached please find a copy of pcap file from S1 interface and a copy of log file. I also configure AuC database file in the following picture. Thanks

[image]

[corenet_20191114.log](https://github.com/P1sec/pycrate/files/3844327/corenet_20191114.log)
[corenet_S1AP_20191114.zip](https://github.com/P1sec/pycrate/files/3844336/corenet_S1AP_20191114.zip)

LucasCSTI, 2019-11-14:
I do another test and have tried different 2G algorithm (Milenage, comp128v1, comp128v2 or comp128v3), and figure out that my test sim only support XOR algorithm. Thanks

[corenet-S1AP_20191114v2.log](https://github.com/P1sec/pycrate/files/3845360/corenet-S1AP_20191114v2.log)
[corenet_S1AP_20191114v2.zip](https://github.com/P1sec/pycrate/files/3845361/corenet_S1AP_20191114v2.zip)

p1-bmu, 2019-11-14:
So, should I understand that there is no crash anymore (something related to a PLMN field being mis-encoded) ? Did you do something wrong to have this crash ? Did it disappear by itself, or did you change something ? Sorry, but I really fail to understand what was your initial issue. 
Regarding your authentication issue, you need to have properly configured USIM to attach UEs over LTE: have a look at sysmoUSIM at sysmocom-shop.

LucasCSTI, 2019-11-14:
Yes ,it won't crash after I configure AuC database file properly. Thanks

closed 2019-11-14 by LucasCSTI