P1sec/pycrate #68: Add XOR support for UE Auth, then it crash after SMC procedure

LucasCSTI, 2019-11-20:
I add XOR test algorithm to ServerAuC.py ([ServerAuC.zip](https://github.com/P1sec/pycrate/files/3867919/ServerAuC.zip)) with new AuC database ([AuC_20191120.zip](https://github.com/P1sec/pycrate/files/3867982/AuC_20191120.zip)) for corenet-46666.py ([corenet-46666.zip](https://github.com/P1sec/pycrate/files/3867947/corenet-46666.zip))

        if ALG2 == 9:
            # pack SQN from integer to a 48-bit buffer
            SQNb = b'\0\0' + pack('>I', SQN)
            #
            if RAND is None:
                RAND = genrand(16)
            # compute XOR functions
            XDOUT    = xor_buf( K, RAND )
            XRES     = XDOUT[0:8]
            CK       = XDOUT[1:16] + XDOUT[0]
            IK       = XDOUT[2:16] + XDOUT[0:2]
            AK       = XDOUT[3:11]
            CDOUT    = SQNb + AMF
            MAC_A    = xor_buf(XDOUT[0:8], CDOUT)
            SQN_X_AK = xor_buf( SQNb, AK )
            AUTN     = SQN_X_AK + AMF + MAC_A
        else:

Then my UE can pass authentication. (pcap file for S1AP: [corenet_S1AP_20191120v2.zip](https://github.com/P1sec/pycrate/files/3867957/corenet_S1AP_20191120v2.zip))

[image]

However, it crash again after SMC procedure.  (log file for corenet: [corenet_20191120v2.log](https://github.com/P1sec/pycrate/files/3867960/corenet_20191120v2.log))

[image]

-------------------------------------------------------------------------------------------------------
The following procedure employs bit wise modulo 2 addition ("XOR").
The following convention applies:
In all data transfer the most significant byte is the first byte to be sent; data is represented so that the left most bit is the most significant bit of the most significant byte.

[image]

mitshell, 2019-11-20:
From the stacktrace, it seems the application is failing at encoding a PLMN value as part of the GUTI to be set into the EMMAttachAccept message.

Within your corenet.py file, you have set this:
```
# equivalent PLMNs served
# None or list of PLMNs ['30124', '763326', ...]
CorenetServer.EQUIV_PLMN = 'III' # None
```
What is not a valid list of PLMN: you can try to set a correct value there (None, or list of PLMN). 

On the other hand, I suspect that there could be some issues with working with Python2 (especially when encoding digit string into such PLMN object): did you try running corenet with Python3 ?

LucasCSTI, 2019-11-21:
It won't crash when runing corenet with Python3. I also update ServerAuC.py ([ServerAuC.zip](https://github.com/P1sec/pycrate/files/3872450/ServerAuC.zip)) to support XOR test algorithm for Python3.

        if ALG2 == 9:
            # pack SQN from integer to a 48-bit buffer
            SQNb = b'\0\0' + pack('>I', SQN)
            #
            if RAND is None:
                RAND = genrand(16)
            # compute XOR functions
            XDOUT    = xor_buf( K, RAND )
            XRES     = XDOUT[0:8]
            CK       = XDOUT[1:16] + XDOUT[0:1]
            IK       = XDOUT[2:16] + XDOUT[0:2]
            AK       = XDOUT[3:11]
            CDOUT    = SQNb + AMF
            MAC_A    = xor_buf(XDOUT[0:8], CDOUT)
            SQN_X_AK = xor_buf( SQNb, AK )
            AUTN     = SQN_X_AK + AMF + MAC_A

Then my UE can successfully connect to corenet now. (pcap file for S1AP: [corenet_S1AP_20191121.zip](https://github.com/P1sec/pycrate/files/3872458/corenet_S1AP_20191121.zip))

[image]

closed 2019-11-21 by LucasCSTI

mitshell, 2019-11-21:
When using Python2, you need to set string variable as unicode to have proper encoding, such as:
```
CorenetServer.PLMN = u'46666'
```
But using Python3 seems more straightforward.

Thanks for the proposal regarding the XOR authentication algorithm: I'll check to integrate it into the mainline.