P1sec/pycrate #73: S1AP Decoding Error

suchitsatpathy, 2020-01-23:
The decoding is failing to recognize the IE - ID for example below 

     {
        id 24,
        criticality reject,
        value '0100340043450006300F800AA49539041C606E3427ED14AD4705FBBB003176EF375BF679DE8AD928213DD92DBA729D52B83BBA1C32E5E64D700EE8C11CF8A3588B4F94A8ADE304430034000E060005260F800AA4953902B8406E'H

The prodedure id = 9 . 

in the IE explanation IE-24 = id-E-RABToBeSetupListCtxtSUReq in S1AP-Constants.asn definition .

Below are the errors -

OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
OPEN._from_per: SuccessfulOutcome.protocolIEs._item_.value, unable to retrieve a table-looked up object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object

suchitsatpathy, 2020-01-23:
Update:

Found that i was calling the wrong message type method

S1AP.S1AP_PDU_Descriptions.InitiatingMessage , The correct one was InitiatingMessage . I was able to decode most of the IE except UERadioCapability .

 Do i have to call the RRC or NAS decodes to parse it ?

p1-bmu, 2020-01-27:
Yes, generally, UERadioCapability is just a container with a list of the different technologies radio capabilities useful for managing handovers:
"MS Classmark" 2 and 3 for 2G, "InterRAT Handover Info" for UMTS, "UE EUTRA Capability" for 4G...
You can find an example on how to decode it here: https://github.com/P1sec/pycrate/blob/master/pycrate_corenet/utils.py

closed 2020-01-27 by p1-bmu

suchitsatpathy, 2020-02-13:
This is awesome .

Thank You .

Suchit Satpathy