P1sec/pycrate #77: M3UA support for Variable Length parameter decoding and encoding

slk2k, 2020-04-03:
Hello,

Is there any plan on supporting the various Variable Length Parameters from rfc4666 (section 3,2)?  In looking over SIGTRAN.py, I do not see any definitions to the Variable Length Format messages such as Network Appearance, Routing Context, Protocol Data or Correlation ID from a Transfer Message in section 3.3 (a very common message) found in the MTP3 component of the M3UA message?? 

Thank you for all your work on the project - it does work fantastically!!

mitshell, 2020-04-04:
Hi,
Thanks for your message.

M2UA/M3UA parameters are all in Tag-Length-Value format, if I remember well. They are all decoded as such: https://github.com/P1sec/pycrate/blob/e11b6bd541123a5c623fd629ef61f0b8d0b5a648/pycrate_mobile/SIGTRAN.py#L461
And tags value are associated with their meaning thanks to this dictionnary: https://github.com/P1sec/pycrate/blob/e11b6bd541123a5c623fd629ef61f0b8d0b5a648/pycrate_mobile/SIGTRAN.py#L358

I never had any need to implement specific structures for SIGTRAN parameter's value, but any contributions on this is welcome ! Feel free to contribute.

slk2k, 2020-04-05:
Thank you for the response!!  I was able to parse into the TLV, but was looking for the structs within the V (Value) to break it down even further in order to get at the next level of the protocol payload.

p1-bmu, 2020-04-29:
Actually, the structures for each M3UA parameter and message are defined within RFC 4666, from section 3.3 to 3.8. They could be implemented within SIGTRAN.py.

p1-bmu, 2023-10-23:
Finally, formats from RFC 4666 have been implemented in [pycrate_mobile/M3UA.py](https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/M3UA.py). I guess we can close this.

closed 2023-10-23 by p1-bmu