P1sec/pycrate #78: Strange behaviour when encoding RRC message with wrong type

FizzyJelly, 2020-04-20:
Hi,

I have come across a questionable behaviour when encoding RRC message with wrong type.
The message type was BCCH-DL-SCH-Message, but somebody have mistaken it for BCCH-BCH-Message.

```python
from pycrate_asn1dir import RRCLTE

defs = RRCLTE.EUTRA_RRC_Definitions
msg = defs.BCCH_BCH_Message  # should be BCCH_DL_SCH_Message

msg.from_jer('{"message": {"c1": {"systemInformation": {"criticalExtensions": {"systemInformation-r8": {"sib-TypeAndInfo": [{"sib2": {"ac-BarringInfo": {"ac-BarringForEmergency": false, "ac-BarringForMO-Data": {"ac-BarringFactor": "p95", "ac-BarringForSpecialAC": "00", "ac-BarringTime": "s8"}, "ac-BarringForMO-Signalling": {"ac-BarringFactor": "p95", "ac-BarringForSpecialAC": "00", "ac-BarringTime": "s8"}}, "freqInfo": {"additionalSpectrumEmission": 1, "ul-Bandwidth": "n100", "ul-CarrierFreq": 19400}, "radioResourceConfigCommon": {"bcch-Config": {"modificationPeriodCoeff": "n8"}, "pcch-Config": {"defaultPagingCycle": "rf64", "nB": "quarterT"}, "pdsch-ConfigCommon": {"p-b": 1, "referenceSignalPower": 8}, "prach-Config": {"prach-ConfigInfo": {"highSpeedFlag": false, "prach-ConfigIndex": 3, "prach-FreqOffset": 4, "zeroCorrelationZoneConfig": 12}, "rootSequenceIndex": 12}, "pucch-ConfigCommon": {"deltaPUCCH-Shift": "ds2", "n1PUCCH-AN": 36, "nCS-AN": 0, "nRB-CQI": 4}, "pusch-ConfigCommon": {"pusch-ConfigBasic": {"enable64QAM": false, "hoppingMode": "interSubFrame", "n-SB": 1, "pusch-HoppingOffset": 6}, "ul-ReferenceSignalsPUSCH": {"cyclicShift": 0, "groupAssignmentPUSCH": 0, "groupHoppingEnabled": false, "sequenceHoppingEnabled": false}}, "rach-ConfigCommon": {"maxHARQ-Msg3Tx": 5, "powerRampingParameters": {"powerRampingStep": "dB2", "preambleInitialReceivedTargetPower": "dBm-96"}, "preambleInfo": {"numberOfRA-Preambles": "n40", "preamblesGroupAConfig": {"messagePowerOffsetGroupB": "dB10", "messageSizeGroupA": "b144", "sizeOfRA-PreamblesGroupA": "n32"}}, "ra-SupervisionInfo": {"mac-ContentionResolutionTimer": "sf64", "preambleTransMax": "n10", "ra-ResponseWindowSize": "sf10"}}, "soundingRS-UL-ConfigCommon": {"release": null}, "ul-CyclicPrefixLength": "len1", "uplinkPowerControlCommon": {"alpha": "al04", "deltaFList-PUCCH": {"deltaF-PUCCH-Format1": "deltaF0", "deltaF-PUCCH-Format1b": "deltaF1", "deltaF-PUCCH-Format2": "deltaF0", "deltaF-PUCCH-Format2a": "deltaF0", "deltaF-PUCCH-Format2b": "deltaF0"}, "deltaPreambleMsg3": 1, "p0-NominalPUCCH": -116, "p0-NominalPUSCH": -80}}, "timeAlignmentTimerCommon": "sf10240", "ue-TimersAndConstants": {"n310": "n10", "n311": "n1", "t300": "ms400", "t301": "ms400", "t310": "ms2000", "t311": "ms3000"}}}]}}}}}}')
uper = msg.to_uper()
print(uper)

msg.from_uper(uper)
```

When executed it produces following output:

```shell
b'\x00'
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    msg.from_uper(uper)
  File "./asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj.py", line 1195, in from_uper
    self._from_per(char)
  File "./asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_construct.py", line 800, in _from_per
    Comp._from_per(char)
  File "./asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_construct.py", line 800, in _from_per
    Comp._from_per(char)
  File "./asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_str.py", line 470, in _from_per
    buf = char.get_bytes(ldet)
  File "./asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_core/charpy.py", line 375, in get_bytes
    raise(CharpyErr('bitlen overflow: {0}, max {1}'\
pycrate_core.charpy.CharpyErr: bitlen overflow: 8, max 2
```

My question is: shouldn't the encoding method 'to_uper' throw an error? Or even the 'from_jer' method as the mistaken message types are quite different.
It makes sense that if it is not possible to decode the message back then it should not have been possible to encode the message in the first place.

I am using Python 3.8 and pycrate 0.4.
Cheers

p1-bmu, 2020-04-20:
Thanks for your feedback.

1) I must admit that checks on value set for encoding maybe bypassed in certain cases (this can be hard to set the boundary on what to check and what not when providing a library), especially in the JER case.

2) JER is barely defined, especially for BIT STRING and everything related to bit-level operations (hence with all protocols using PER), and I can only discourage you to use it.

3) You can use the _safechk_val() (e.g. here: msg._safechk_val(msg._val)) method explicitely on ASN.1 objects to check for value correctness after setting them.

4) I will check to potentially enhance this value checks at setting / encoding, especially for JER, but this is not a high priority for me now. Any contributions welcome :)

Regards
Benoit

FizzyJelly, 2020-04-22:
So I tried out proposed '_safechk_val' method and it generally does well.

However in some cases when you happen to misspell a key a mysterious ValueError pops up when 'from_jer' is called. That succesfully prevents you from finding out what happened.

```shell
Traceback (most recent call last):
  File "test.py", line 8, in <module>
    msg.from_jer(json.dumps(
  File "~/asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj.py", line 1739, in from_jer
    self._from_jval(val)
  File "~/asn1-remote-decoder/venv/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_construct.py", line 1134, in _from_jval
    for ident, comp_val in val_cp:
ValueError: too many values to unpack (expected 2)
```
When I dug into the case, I noticed that the line 1134 in pycrate_asn1rt.asnobj_construct.py has a bug - lack of '.items()' when iterating over a dictionary.

```python
1133            if val_cp:
1134                for ident, comp_val in val_cp:
1135                   self._val['_ext_%s' % ident] = comp_val
```
When this is fixed, json decoding finishes and you can actually use the _safechk_val() method and get more appropriate error.

p1-bmu, 2020-04-23:
I just fixed this here: https://github.com/P1sec/pycrate/commit/bdb01cc9e7f03437f37388acb389e3f55e487f6e
Thanks for your analysis

p1-bmu, 2020-04-27:
I just pushed https://github.com/P1sec/pycrate/commit/2d95a525a00815e55d81f8705e6959759d636417
This will check for complete and correct value when decoded from JER. I reviewed other JER decoding routines, which are looking right at ensuring values are correct otherwise. So this sould solve your initial issue without explicit addition.

closed 2020-04-29 by p1-bmu

FizzyJelly, 2020-04-29:
Thank you for reviewing this issue,

Best regards