P1sec/pycrate #80: Does the program pars LTE NAS EMM attach request message incorrectly ?

ccwcm91, 2020-04-29:
Dear Sir:

I tried to parsing the real message from my device
"0741120BF664F629800001D2906DA605F070C04019002A0213D011D127238080211001000010810600000000830600000000000D00000A000005000010000011005264F6292E185C0A00**3103E5E03E**1364F629283411034F18A440080402600400021F025D0103E0C110021E00"

It returns decoded information
 ### MSNetCap ###
  <T : 49>
  <L : 3>
  <ms_network_capability_value_part (CSN1List): [
   <(CSN1Ref): <gea1_bits (CSN1Bit): 1>>
   <sm_capabilities_via_dedicated_channels (CSN1Bit): 1>
   <sm_capabilities_via_gprs_channels (CSN1Bit): 1>
   <ucs2_support (CSN1Bit): 0>

   <ss_screening_indicator (CSN1Bit): 1>
   <solsa_capability (CSN1Bit): 0>
   <revision_level_indicator (CSN1Bit): 1>

   <pfc_feature_mode (CSN1Bit): 1>
   <(CSN1Ref): <extended_gea_bits (CSN1List): [
    <gea_2 (CSN1Bit): 1>
    <gea_3 (CSN1Bit): 1>
    <gea_4 (CSN1Bit): 0>
    <gea_5 (CSN1Bit): 0>
    <gea_6 (CSN1Bit): 0>
    <gea_7 (CSN1Bit): 0>]>>
   <lcs_va_capability (CSN1Bit): 0>

   <ps_inter_rat_ho_from_geran_to_utran_iu_mode_capability (CSN1Bit): 0>
   <ps_inter_rat_ho_from_geran_to_e_utran_s1_mode_capability (CSN1Bit): 0>
   <emm_combined_procedures_capability (CSN1Bit): 1>
   <isr_support (CSN1Bit): 1>
   <srvcc_to_geran_utran_capability (CSN1Bit): 1>
   <epc_capability (CSN1Bit): 1>
   <nf_capability (CSN1Bit): 1>
   **<geran_network_sharing_capability (CSN1Bit): 0>**]>
I know the fomat is form 3GPP 24.008  Table 10.5.145/3GPP TS 24.008 MS network capability information element.
However, I wonder if the last bit is "GERAN network sharing capability", since the other parser shows that the last one should be "Spare bit."

Actually, I am not sure what is each bit represents for, when the length is 3 or 4 or 5 or ...or 10.

Thanks

p1-bmu, 2020-04-29:
The last nibble of your IE value is 0xE, which is 0b1110 in binary.
This seems legit to map to the following csn.1 bits as such:
```
 <srvcc_to_geran_utran_capability (CSN1Bit): 1>
 <epc_capability (CSN1Bit): 1>
 <nf_capability (CSN1Bit): 1>
 <geran_network_sharing_capability (CSN1Bit): 0>
```
Regarding your "other parser", I don't know what it is, where it is... Maybe it's simply wrong ?

ccwcm91, 2020-04-29:
Consider two cases:
1. MS network capability contains below fields:
GEA1, SM capabilities via dedicated channels, ..., NF capability
Totally 23 bits

2. MS network capability contains below fields:
GEA1, SM capabilities via dedicated channels, ..., NF capability **and GERAN network sharing capability=0**
Totally 24 bits

For case 1, the spare bit should be added at last then we get "3103E5E03E"
For case 2, 
(1)the spare bit is still needed? We still sent "3103E5E03E"?
(2)the spare bits are sent, add the length to 4, we sent "3104E5E03E00"??

Maybe the sparse bits  represents all other fields are set to 0, so case 1 and case 2(1) are the same meaning.

Thanks

p1-bmu, 2020-04-29:
I don't have any answer unfortunately. Maybe the 3GPP standards has some guidance on this.

closed 2020-05-12 by p1-bmu