P1sec/pycrate #81: Objects are mutable and retain values out side the function

suchitsatpathy, 2020-05-07:
Hi;
I created a function to decode various LTE RRC messages . Created a mapping dictionary to assign objects based on message type . When i executed it multiple times , I realized that the objects are retaining the values from previous function call . 

 In order to overcome this situation, I used copy.deepcopy . It seems it is adding significant processing time and computing . Is there any initialization method or object reset function available which is efficient ?

p1-bmu, 2020-05-07:
Hello, thanks for your feedback.

Most of the top-level ASN.1 objects in 3GPP standards, which correspond to signaling messages / PDUs, are huge (with thousands of sub-objects). There is no _clone()_ method for ASN.1 objects, as it would be equivalent to do a deepcopy and would require a lot of processing at each call for those complex object.

To me, the easiest and lightweight way to keep track of ASN.1 objects' values is to only retain their values (available under their __val_ attribute or returned when calling the object).

suchitsatpathy, 2020-05-11:
Hi ;

Thanks for the response . So , does the ._val resets in each function call ? Do you have any example that you can share . I just do not want to mess up the out put after each function call .

BR

Suchit

Suchit Satpathy

p1-bmu, 2020-05-12:
ASN.1 object's value never gets reset, it is just overwritten when required (as there is no need for reset). You can find an example on how 3G-4G-5G PDUs are read and processed in corenet, e.g.: 
https://github.com/P1sec/pycrate/blob/ffe38db8dd5626e189957b4b04d1046a19f329c9/pycrate_corenet/Server.py#L665
You can see that, as soon as the PDU is decoded, the value is collected for further processing whereas the lock on the ASN.1 object is released (for safe multi-threading).

suchitsatpathy, 2020-05-23:
Hi Mich ;

I am struggling little bit on how to insert the UE Capability Information decode in the middle of the S1AP object . Do you have any examples that i can follow .

Thanks in advance

Suchit Satpathy

https://github.com/P1sec/pycrate/blob/ffe38db8dd5626e189957b4b04d1046a19f329c9/pycrate_corenet/Server.py#L665

p1-bmu, 2020-05-25:
Please open a new issue for this independent topic.
You should also close the issues to which an answer / solution was provided.
Thanks

closed 2020-05-28 by p1-bmu