P1sec/pycrate #82: X2AP message decode providing error

suchitsatpathy, 2020-05-14:
Hi ;

I was trying to decode the following x2ap message and saw the warning messages . 

I event downloaded the latest version of the spec [3GPP TS 36.423 V16.0.0] and used the 'python -m pycrate_asn1c.asnproc' command to compile it again . 

Is there any property for the ASN 0bject to verify what is the spec version and is there any option to suppress these warnings.

Msg: 001c4030000004006f0002078300cf00048001fc0700d64003100100a8dd40140000021041400804de796c949ef6211042400100

OutPut :
OPEN._from_per: InitiatingMessage.protocolIEs._item_.value, unable to retrieve a table-looked up object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
('initiatingMessage : {\n'
 '  procedureCode 28,\n'
 '  criticality ignore,\n'
 '  value SgNBReconfigurationComplete: {\n'
 '    protocolIEs {\n'
 '      {\n'
 '        id 111,\n'
 '        criticality reject,\n'
 '        value UE-X2AP-ID: 1923\n'
 '      },\n'
 '      {\n'
 '        id 207,\n'
 '        criticality reject,\n'
 '        value SgNB-UE-X2AP-ID: 130055\n'
 '      },\n'
 '      {\n'
 '        id 214,\n'
 '        criticality ignore,\n'
 '        value ResponseInformationSgNBReconfComp: success-SgNBReconfComp : {\n'
 "          meNBtoSgNBContainer '00'H\n"
 '        }\n'
 '      },\n'
 '      {\n'
 '        id 43229,\n'
 '        criticality ignore,\n'
 "        value '0000021041400804DE796C949EF6211042400100'H\n"
 '      }\n'
 '    }\n'
 '  }\n'
 '}')

p1-bmu, 2020-05-14:
I guess the message you provided is not corresponding to a `X2AP_PDU_Descriptions.X2AP_PDU`, maybe a substructure of it, or a message with some proprietary structures ?
Where does it come from ? What do you expect as a type of message and content ?

Regarding the versioning of ASN.1 specifications, you can check the OID of a module (when defined), by calling the `_oid_` attribute:
```
In [12]: X2AP_PDU_Descriptions._oid_
Out[12]: [0, 4, 0, 0, 21, 3, 2, 1, 0]
```
On the other side, I am not sure those OID set for 3GPP RAN protocols follow the exact 3GPP TS versioning scheme.

suchitsatpathy, 2020-05-14:
Thanks Ben;

you are right . I used wireshark to decode it and it looks unknown . With that , it is possible to disable these warnings ? So the output is not going to show these messages .

It came from one of the debug print out from one RAN vendor . I think they might have some proprietary information in there . I will just strip these bits as it is same for all X2 messages.

[cid:73799cef-523f-4c2d-9f14-921bb8a301bc]
Suchit Satpathy

p1-bmu, 2020-05-15:
This should silent the prints related to ASN.1 encoding / decoding:
```
In [1]: from pycrate_asn1rt.asnobj import ASN1Obj

In [2]: ASN1Obj._SILENT = True
```

closed 2020-05-28 by p1-bmu