P1sec/pycrate #84: MeNBtoSgNBContainer - Decoding capability .

suchitsatpathy, 2020-05-23:
Hi ;

I am trying to decode the following X2 PDU 👍 
001b00838e000009006f0002078300f800051c000e000000cb0020ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff00cc000a184c096c70604c096c7000cd001f0000d1001a0a2000c0072d01f02a6d0fd4ea69f4fa03e00ac612798083809d00ce0082f982f718988169210229ce400000028ebc0606000002809049a3000481a0603100d00980406228040530805502c46d618c21a0c54083e500892d931541439f60478c73e618f28581c0e1e04fc0000003f00000000a00e0540b40f78e3087000a8f3f140453ed98aa9041e3c471e00438820c22a051fbf90202e3718120180a816826f1c610e00151e7e2808a7db31552083c788e3c0087104184540a3f7f20405c6e30240300020058030a80242080108c062023727802203014008000008818268000e7cc31e50b0302801000802320304d4000438820c22a0407c010040000040280000530058115ba400410000075d240400004000040150288aed40104800d028010a16000102030406070b0c11121318191b252627282d2f4146400542041f0220900834120e230000002220980802412e230000012201002c01854012104008460310118140f40a24e9d3b639f0e24e9d3b639f0c00044000100000080202004008800020100010000400801000907e568262acbde3802000003fcd4ff816e0c814f5c40b04431c55fc8120dc4e81fa6400018c304108502041e6d80835ba74ee20b1a0fcad01181f5d00000033636c9158b11b82010000008400d0146c0118219c0000003371b648b104400e30350f6643182f18437106fb34d0b163a420a08dc575c004048092616d40247fc0000000004001041c0dc2108003c4459483832081001102009000080a0202c3113a80038e0c00ca00080099aaa2400600082061300001990a0181000cc9500c100066528060c00332d40308001998a0100601821349a47493400800c10ec610dc41bc7fe851ce6293940a0802015c000200004000000000422b5515810c0004210003880b1c30060202208001c4058e18030201084000e202c7100181808820007101638800c108a950007101638204110030001020300400403809c001c704000000801000201400880a000802f000416e208208208208784ff907f8198cc00100000002a8000f00007e56820084000814230000102240ac0fc000000112928091409289f8fc5174d870116000100011700080013200614d09010a8dd40140000021041400804de796c949ef6211042400100

The decoder is decoding it well . But it is not decoding , **MeNBtoSgNBContainer**' . For that was i was trying to use the internode definitions but looks like it is providing me some errors.

 Is there any module exists to decode this ?

p1-bmu, 2020-05-25:
As you can read from the TS 36.423 specification on X2AP, the `MeNB to SgNB Container` IE can include different element from the TS 38.331 (NR RRC), depending of the X2AP message type.
Each case is defined textually within section 9 from TS 36.423, and is not handled as the ASN.1 level, where this container is simply defined as an OCTET STRING.

Therefore, it is up to your application to pick up the right ASN.1 object from the module `pycrate_asn1dir.RRCNR` depending of the X2 message type, and decode it explicitly.

p1-bmu, 2020-05-25:
Here is the decoding of your specific message:
```
In [27]: from pycrate_asn1dir.RRCNR import *

In [28]: c = NR_InterNodeDefinitions.CG_ConfigInfo

In [29]: buf = p.get_val_at(('initiatingMessage', 'value', 'SgNBAdditionRequest', 'protocolIEs', 5, 'value', 'MeNBtoSgNBContainer'))

In [30]: c.from_uper(buf)

In [31]: print(c.to_asn1())
{
  criticalExtensions c1 : cg-ConfigInfo : {
    ue-CapabilityInfo UE-CapabilityRAT-ContainerList: {
      {
        rat-Type nr,
        ue-CapabilityRAT-Container 'E720000001475E0303000001404824D1800240D03018806804C0203114020298402A816236B0C610D062A041F2804496C98AA0A1CFB023C639F30C7942C0E070F027E0000001F80000000500702A05A07BC718438005479F8A0229F6CC554820F1E238F0021C4106115028FDFC810171B8C0900C0540B41378E3087000A8F3F140453ED98AA9041E3C471E00438820C22A051FBF90202E371812018001002C0185401210400846031011B93C0110180A00400000440C13400073E618F28581814008004011901826A00021C41061150203E00802000002014000029802C08ADD20020800003AE9202000020000200A8144576A0082400681400850B0000810182030385860889098C0C8D929313941697A0A3200'H
      },
      {
        rat-Type eutra-nr,
        ue-CapabilityRAT-Container '8107C08824020D048388C000000888260200904B88C000004880400B006150048410021180C40460503D02893A74ED8E7C3893A74ED8E7C3000110000400000200808010022000080400040001002004'H
      }
    },
    candidateCellInfoListMN {
      {
        ssbFrequency 129370,
        measResultNeighCellListNR {
          {
            physCellId 197,
            measResult {
              cellResults {
                resultsSSB-Cell {
                  rsrp 47,
                  rsrq 60
                }
              }
            }
          }
        }
      }
    },
    configRestrictInfo {
      allowedBC-ListMRDC {
        {
          bandCombinationIndex 1,
          allowedFeatureSetsList {
            128
          }
        }
      },
      powerCoordination-FR1 {
        p-maxNR-FR1 23
      },
      servCellIndexRangeSCG {
        lowBound 7,
        upBound 31
      }
    },
    sourceConfigSCG RRCReconfiguration: {
      rrc-TransactionIdentifier 0,
      criticalExtensions rrcReconfiguration : {
        secondaryCellGroup CellGroupConfig: {
          cellGroupId 1,
          rlc-BearerToAddModList {
            {
              logicalChannelIdentity 1,
              servedRadioBearer drb-Identity : 3,
              rlc-Config am : {
                ul-AM-RLC {
                  sn-FieldLength size18,
[... 813 lines cut ...]
              pdcp-SN-SizeDL len18bits,
              headerCompression notUsed : NULL
            },
            moreThanOneRLC {
              primaryPath {
                cellGroup 0
              },
              ul-DataSplitThreshold infinity
            },
            t-Reordering ms200
          }
        }
      },
      securityConfig {
        securityAlgorithmConfig {
          cipheringAlgorithm nea3
        },
        keyToUse secondary
      }
    }
  }
}
```

Note that the NR RRC message is encoded with Unaligned PER, and not Aligned PER as X2AP.

suchitsatpathy, 2020-05-25:
Thanks a lot Mich for the explanation .

I was trying to do this yesterday and selected the CG_ConfigInfo object , based on the decoded message using wireshark . I could not figure out how to see if it is a CG_ConfigInfo or some other type .

BR

Suchit Satpathy

closed 2020-05-28 by p1-bmu