P1sec/pycrate #85: Decoding Help for Internode definitions

suchitsatpathy, 2020-05-25:
I am trying to decode the messages for S1AP and X2AP messages , where there are UE capability and other containers included . I can decode them separately using different funtions . However , I am trying to merge all the segments together . Below are the steps I am trying to perform .
  1: Check if a specific container is present in the message 
  2: Decode the message 
  3: set the value in the S1AP message RAT capability type to the decoded value.

p1-bmu, 2020-05-26:
What is the issue you have with pycrate ??

suchitsatpathy, 2020-05-26:
Hi Mich ;

It is rather an enhancement request to decode the subcontainer and update the master container .

BR

Suchit Satpathy

p1-bmu, 2020-05-27:
Dear Suchit.

As you can understand, I am not your developer. Therefore I will not develop specific features for your need. Moreover, I don't understand clearly what you are expecting.
Pycrate is an open-source software, it is open to external contributions, in multiple forms ; see https://github.com/P1sec/pycrate#contributing. If you have specific need and that you succeed to address them with a specific development, you can provide it back through a pull request, so that all pycrate's users can benefit from it.

Regarding the decoding of UE radio capabilities, you can take this part of pycrate_corenet as an example:
https://github.com/P1sec/pycrate/blob/ffe38db8dd5626e189957b4b04d1046a19f329c9/pycrate_corenet/utils.py#L227

Best Regards,
Benoit

suchitsatpathy, 2020-05-27:
Hi Mich,

 I really appreciate and thankful to you for your pycrate initiative and your support.

 About the enhancements I was communicating to you;  I am extreamly sorry , I did not mean to steer or personalize the development . I just wanted to be a part of this awesome initiative and contribute as much i can by testing, suggesting enhancements etc. I came across the contribution page just after I replied to your email. I think the pull option is a right way to follow .
 I am also writing a script to help me debug the system issues . Your initiative really helped me a lot and I am learning from this . Hope , you will not mind if I ask you questions if I have difficulty decoding a 3GPP messgae.

BR

Suchit Satpathy

https://github.com/P1sec/pycrate/blob/ffe38db8dd5626e189957b4b04d1046a19f329c9/pycrate_corenet/utils.py#L227