P1sec/pycrate #86: Decoding 5G NAS pdu

suchitsatpathy, 2020-06-03:
I am trying to decode the following NAS-PDU , I am getting error : 97

'value': ('InitialUEMessage',
                                                                                  {'protocolIEs': [{'criticality': 'reject',
                                                                                                    'id': 85,
                                                                                                    'value': ('RAN-UE-NGAP-ID',
                                                                                                              11918)},
                                                                                                   {'criticality': 'reject',
                                                                                                    'id': 38,
                                                                                                    'value': ('NAS-PDU',
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff'
                                                                                                              b'\xff\xff\xff\xff')},

Is it something I am doing wrong or may be there is a better way . 

Below is what I am doing -

nasPDUBytes = b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
msg,er = NAS.parse_NAS_MO(nasPDUBytes)
print(err)
#print(nas_Msg)
print(msg)

p1-bmu, 2020-06-04:
Do you really think b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff' is a valid NAS message ??

I don't think so, hence error 97 (https://github.com/P1sec/pycrate/blob/ffe38db8dd5626e189957b4b04d1046a19f329c9/pycrate_mobile/TS24501_IE.py#L267): Message type non-existent or not implemented.

closed 2020-06-04 by p1-bmu

suchitsatpathy, 2020-06-04:
Hi Mich;

You are right . I realized that it is ciphered .  Then I tried one payload with no ciphering and it worked .

Is there any existing module you have created to decipher if the key is available ? Just a thought ..

Thanks

Suchit Satpathy