P1sec/pycrate #87: RRCLTE:SIB1 decoding fails when the SEQUENCE SIZE is 0

renmnair, 2020-06-12:
I am using this library to encode and decode the RRCLTE(NbIoT) messages. I am seeing error as below when decoding SIB1.
Steps:
```
from pycrate_asn1dir import RRCLTE
from binascii import unhexlify, hexlify
sib1 = RRCLTE.NBIOT_RRC_Definitions.BCCH_DL_SCH_Message_NB
sib1.from_uper(unhexlify('6040008090d20004345a20500ba010300e00'))
print(sib1.to_asn1())
```

and I get the error as below:

`    print(sib1.to_asn1())
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj.py", line 1170, in to_asn1
    return self._to_asn1()
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 645, in _to_asn1
    val.append('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 123, in _to_asn1
    ret = '%s : %s' % (ident, self._cont[ident]._to_asn1())
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 123, in _to_asn1
    ret = '%s : %s' % (ident, self._cont[ident]._to_asn1())
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 645, in _to_asn1
    val.append('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 2064, in _to_asn1
    val.append('  %s,\n' % self._cont._to_asn1().replace('\n', '\n  '))
  File "/home/rnair/Downloads/pycrate-master/pycrate_asn1rt/asnobj_construct.py", line 645, in _to_asn1
    val.append('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 1: ordinal not in range(128)
`

**val.append('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))** fails when 
 when there is 0 items in the sib_MappingInfo_r13.

[image]

As per 3GPP specification 36.331, ASN.1 defenition is as:
[image]

Can you please take a look into this issue?

p1-bmu, 2020-06-15:
I checked you case both on Windows (python 3.6) and on Ubuntu (python 2.7 and 3.6) and the issue only exists in Python2.7. I can only encourage you to migrate to Python3, as Python2 is not supported anymore since few months.

Here with Python3:
```
In [4]: m.from_uper(unhexlify('6040008090d20004345a20500ba010300e00'))

In [5]: m._val
Out[5]: 
{'message': ('c1',
  ('systemInformationBlockType1-r13',
   {'cellAccessRelatedInfo-r13': {'cellBarred-r13': 'notBarred',
     'cellIdentity-r13': (27447554, 28),
     'intraFreqReselection-r13': 'allowed',
     'plmn-IdentityList-r13': [{'cellReservedForOperatorUse-r13': 'notReserved',
       'plmn-Identity-r13': {'mcc': [0, 1, 2], 'mnc': [3, 4]}}],
     'trackingAreaCode-r13': (2, 16)},
    'cellSelectionInfo-r13': {'q-QualMin-r13': -23, 'q-RxLevMin-r13': -70},
    'freqBandIndicator-r13': 5,
    'hyperSFN-MSB-r13': (0, 8),
    'p-Max-r13': 10,
    'schedulingInfoList-r13': [{'si-Periodicity-r13': 'rf512',
      'si-RepetitionPattern-r13': 'every2ndRF',
      'si-TB-r13': 'b256',
      'sib-MappingInfo-r13': []}],
    'si-WindowLength-r13': 'ms960',
    'systemInfoValueTagList-r13': [0]}))}

In [6]: print(m.to_asn1())
{
  message c1 : systemInformationBlockType1-r13 : {
    hyperSFN-MSB-r13 '00'H,
    cellAccessRelatedInfo-r13 {
      plmn-IdentityList-r13 {
        {
          plmn-Identity-r13 {
            mcc {
              0,
              1,
              2
            },
            mnc {
              3,
              4
            }
          },
          cellReservedForOperatorUse-r13 notReserved
        }
      },
      trackingAreaCode-r13 '0002'H,
      cellIdentity-r13 '1A2D102'H,
      cellBarred-r13 notBarred,
      intraFreqReselection-r13 allowed
    },
    cellSelectionInfo-r13 {
      q-RxLevMin-r13 -70,
      q-QualMin-r13 -23
    },
    p-Max-r13 10,
    freqBandIndicator-r13 5,
    schedulingInfoList-r13 {
      {
        si-Periodicity-r13 rf512,
        si-RepetitionPattern-r13 every2ndRF,
        sib-MappingInfo-r13 { },
        si-TB-r13 b256
      }
    },
    si-WindowLength-r13 ms960,
    systemInfoValueTagList-r13 {
      0
    }
  }
}
```

I will try to fix the issue with Python2, but this is only a low priority to me.

p1-bmu, 2020-06-15:
I commited https://github.com/P1sec/pycrate/commit/ed0e6b6bf3c87c59ac290a232679ec4d2783133d, that should solve this issue.

renmnair, 2020-06-16:
Thanks, I have updated to Python 3 and it works.

closed 2020-06-16 by renmnair