P1sec/pycrate #90: error on decode ussd hex

parsibox, 2020-06-25:
hi
i want decode this hex:
please help me
```
000ded2a4d80000d608973ee0800450000903cb0000080847895c01c9776c01c94f50ded0deda3e011f54484baf20003006d1357a7a6000102ab0000000501000b010000005d000001ba000001bb008302400090090103050902420804430100083962374804000204906b1a2818060700118605010101a00d600ba1090607040000010013026c13a11102010102013b300904010f04042a1c6ed4000000
```

and

```
000400010006e83935c3e312000008004500010c00004000ff84f880c0a8004dc0a8004f0b590b59d1dbc0a1826e68ff000300ec3827a6da000100010000000301000101000000dc02000008000000010006000800000002021000c4000007e4000003f203020300090003070b0443f203070443e40706a46281a14804040000006b3d283b060700118605010101a030602ea109060704000001001302be21281f060704000001010101a014a0128007191989400040448107910000000000006c5aa15802010102013d30500401010443d618681866b341e3f79c0e7281846136483a77c5643397cce602818661361b44aecbc3f4f4dbad03dd40d3f218444c87d920552c16038d40f437688c2e8fd72071980d8006199989080010
```
my code is 

```
#!/usr/bin/env python3
import sys
import time
from test import test_pycrate
from binascii import hexlify, unhexlify
from pycrate_asn1dir import TCAP_MAPv2v3
M = TCAP_MAPv2v3.TCAP_MAP_Messages.TCAP_MAP_Message
M.from_ber(unhexlify('000ded2a4d80000d608973ee0800450000903cb0000080847895c01c9776c01c94f50ded0deda3e011f54484baf20003006d1357a7a6000102ab0000000501000b010000005d000001ba000001bb008302400090090103050902420804430100083962374804000204906b1a2818060700118605010101a00d600ba1090607040000010013026c13a11102010102013b300904010f04042a1c6ed4000000'))
print(M.to_asn1())
```

but i get this error :

```
F:\source\python\Digits-Classification-master\venv\Scripts\python.exe F:/source/python/ss7/pycrate-master/test.py
init_modules: different OID objects (locationCancellation-v2, locationCancellationContext-v2) with same OID value (0, 4, 0, 0, 1, 0, 2, 2)
init_modules: different OID objects (locationCancellation-v1, locationCancellationContext-v1) with same OID value (0, 4, 0, 0, 1, 0, 2, 1)
Traceback (most recent call last):
  File "F:/source/python/ss7/pycrate-master/test.py", line 5, in <module>
    M.from_ber(unhexlify('000ded2a4d80000d608973ee0800450000903cb0000080847895c01c9776c01c94f50ded0deda3e011f54484baf20003006d1357a7a6000102ab0000000501000b010000005d000001ba000001bb008302400090090103050902420804430100083962374804000204906b1a2818060700118605010101a00d600ba1090607040000010013026c13a11102010102013b300904010f04042a1c6ed4000000'))
  File "F:\source\python\ss7\pycrate-master\pycrate_asn1rt\asnobj.py", line 1408, in from_ber
    self._from_ber(char, TLV)
  File "F:\source\python\ss7\pycrate-master\pycrate_asn1rt\asnobj.py", line 1394, in _from_ber
    self._decode_ber_cont(char, tlv)
  File "F:\source\python\ss7\pycrate-master\pycrate_asn1rt\asnobj_construct.py", line 478, in _decode_ber_cont
    .format(self.fullname(), (cl, tval))))
pycrate_asn1rt.err.ASN1BERDecodeErr: TCAP-MAP-Message: invalid CHOICE tag according to the content, (0, 0)

Process finished with exit code 1

```

p1-bmu, 2020-06-25:
I suppose your hex streams are not TCAP-MAP message...

parsibox, 2020-06-25:
this is my pcap file
please see it and help me
[ussd1.zip](https://github.com/P1sec/pycrate/files/4831070/ussd1.zip)

parsibox, 2020-06-25:
and this is wireshark site sample
[gsm_map_with_ussd_string.zip](https://github.com/P1sec/pycrate/files/4831097/gsm_map_with_ussd_string.zip)

parsibox, 2020-06-25:
@p1-bmu  can you help me to decode  gsm_map_with_ussd_string.zip with your code?

parsibox, 2020-06-25:
you are right it is hex streams 
but how can i extract TCAP-MAP from  hex streams  ?

p1-bmu, 2020-06-26:
Check wireshark wiki on how to extract a hex stream part of a pcap record: https://wiki.wireshark.org/
Then check pycrate wiki on how to decode such message: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime

closed 2020-06-26 by p1-bmu

parsibox, 2020-06-26:
i read your wiki more than 5 time
but my problem is how extract  TCAP-MAP from hex streams
i want do this with python code not with wireshark.
if you can please help me to do this

mitshell, 2020-06-26:
Sorry but I can't.

nhanth87, 2020-06-30:
the pcap decap is hardcoded into wireshark code so if you dont use wireshark, you can't decode the message or you can write your old decoder

parsibox, 2020-06-30:
we have hex stream ( we want extract  TCAP-MAP from hex streams  not from pcap file )

parsibox, 2020-06-30:
@nhanth87 some project in java do that ( jss7 project )
but i need it in python

nhanth87, 2020-06-30:
as I told to you,
the wireshark decoded, it's hardcoded in wireshark code so you can't decode your self until you write your own decoder

the jss7 project also hardcoded in their own code

parsibox, 2020-06-30:
jss7 is open source project and have library.
can we use wireshark hardcode in python?
Does it have a library for this?