P1sec/pycrate #93: Choosing between choices for 'operationModeInfo-r13' NBIoT MIB

renmnair, 2020-07-09:
Hi,

I have a query regarding able to choose  between choices from a default configuration.
Keeping one choices as the default is it possible to choose another.
For example,
mib = RRCLTE.NBIOT_RRC_Definitions.BCCH_BCH_Message_NB
mib.from_uper(unhexlify('a0c3c00000'))

```
In [23]: 
    ...: mib = RRCLTE.NBIOT_RRC_Definitions.BCCH_BCH_Message_NB
    ...: mib.from_uper(unhexlify('a0c3c00000'))
    ...: 
    ...: 

In [24]: mib()
Out[24]: 
{'message': {'systemFrameNumber-MSB-r13': (10, 4),
  'hyperSFN-LSB-r13': (0, 2),
  'schedulingInfoSIB1-r13': 3,
  'systemInfoValueTag-r13': 1,
  'ab-Enabled-r13': True,
  'operationModeInfo-r13': ('standalone-r13', {'spare': (0, 5)}),
  'additionalTransmissionSIB1-r15': False,
  'spare': (0, 10)}}
```

As per 3GPP **operationModeInfo-r13** can have different choices.
Is it possible to set operationModeInfo-r13 to a different choice from a default.
For example, it can also be,
`'operationModeInfo-r13': ('guardband-r13', {'rasterOffset-r13': 'khz-7dot5', 'spare': (0, 3)})`

Thanks in Advance,

p1-bmu, 2020-07-09:
Sure, you can change the value as you wish, and re-encode it.
You can read the Wiki for more information on how to manipulate encoded / decoded values: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime, and check the docstrings of ASN.1 objects within the Python interpreter.

renmnair, 2020-07-09:
thanks,

I am just confirming that , I would need to provide the complete structure for this right.
For example, providing just the value 'guardband-r13', doesn't generate the rest of it.

Also one more question,
are you planning for a stable release for pycrate?

p1-bmu, 2020-07-10:
Pycrate can be considered stable for most of its features, maybe except for the `pycrate_corenet` which is quite experimental. Unit testing is used to ensure nothing breaks. I may do a tagged release once a year, but that does not correspond to anything really more stable.

renmnair, 2020-07-13:
thanks for the information.

closed 2020-07-13 by renmnair