P1sec/pycrate #94: ASN1 ITS messages update

coscar68, 2020-07-15:
This is not an issue, but are there any plans to update ITS to support CAM messages (new protocol version 3) according to ETSI EN 302 637-2 V1.4.1 (2019-04)?

p1-bmu, 2020-07-15:
I tried few months ago to compile the set of ASN.1 files provided by ETSI on their forge: https://forge.etsi.org/rep/ITS/ITS_ASN1
I encountered errors related to file encoding, such as:
```ASN1Err: [proc] unable to read spec file EtsiTs102941TypesAuthorization.asn, 'utf-8' codec can't decode byte 0x93 in position 877: invalid start byte```, therefore I gave up.
It seems some of the ASN.1 files on the ETSI forge are not all encoded in UTF-8, but iso-* (windows' style) or some other codecs... This is a pity.

In case you are aware of any repository gathering all ITS ASN.1 files correctly encoded in ascii or utf-8, do not hesitate to tell me.

KenWhitesell, 2020-07-15:
Specifically, the 0x93 is a "left" double-quotation mark - used by some text editors and word processors for making "nicer looking" quote pairs around words. I've run into this may times in different circumstances.
Somewhere around here I've got some code that replaces these types of characters with their standard counterparts. I use it as a filter with files that have these problems. Basically, it's just a search & replace process. You could probably do the same thing with the tr command to clean them up.

p1-bmu, 2020-07-22:
Unfortunately, I don't have time to spend on ITS standards yet. On the other side, if someone provides a PR with an updated ASN.1 directory for ITS, I will be happy to compile / integrate / replace the old one with it.

JNevrly, 2020-09-21:
Continuing discussion from #100 on difficulty of the ITS-related ASN.1 files.

I've tried using [asn1c ](https://github.com/vlm/asn1c)  to compile the ITS Security (TS102941) ASN.1 files, then used the possibility to build a default decoder/encoder utility to make an utility for one of the top-level elements (`RcaCertificateTrustListMessage`), then used this utility to decode a real trust-list message. All worked without problems. This means that also underlying elements from TS103097, which were the subject of the issue #100 were compiled and used correctly.

So yeah, I can report there is at least one open-source project that can handle the ITS-security-level-asn1 mess!

Also, that the compiler can handle OER eventhough it is not mentioned as feature in it's web documentation ... :facepalm:

p1-bmu, 2020-09-22:
Excellent. I still believed that it could not handle parameterization, some CLASS declaration and few other stuff (e.g. APER codec). But it seems the compiler has been completed very well 2 years ago. And it seems APER is supported in a fork (but not in Lev Walkin main repo). This is good news !
Thanks for testing this on your side. I will check how to use it to analyse some of the worst objects within the ITS spec.

closed 2023-02-06 by p1-bmu