P1sec/pycrate #97: Encoding NAS5G messages - syntax help needed

slashTPA, 2020-07-26:
Hi there,

I somehow struggle to instantiate NAS5G messages.  E.g. trying:

>>> **Msg = 5GMMRegistrationAccept()**

gives only back : SyntaxError: **invalid syntax**

I failed to find the right syntax in 5G case (working well in 3G). Any hint would be appreciated. Thx. Do I need to define myself in **TS24501_FGSM.py** this mapping (?) 

#__all__ = [
#    ]

p1-bmu, 2020-07-28:
Hello,
in Python, variables cannot start with an alphanumeric character. Therefore, all "5G" related structures and classes are prefixed "FG", as you can see from the source code, e.g.: https://github.com/P1sec/pycrate/blob/383fd140a60ab5ada688aaec73a122117c879b80/pycrate_mobile/TS24501_FGMM.py#L152

Otherwise, there is no specific difference compared to 2-3-4G NAS protocols, as you can see from the wiki: https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages

slashTPA, 2020-07-28:
Well, still I don't see the relation. If I do a:

Msg, err = parse_NAS5G(unhexlify('2e0501c1ffff91a1')) 
print(Msg.show())
I get a decoding as 5GSMPDUSessionEstabRequest

Running then something like:
Msg = FGSMPDUSessionEstabRequest()
ends up in:
NameError: name 'FGSMPDUSessionEstabRequest' is not defined

while in the 'old' world it worked like:
Msg = MMLocationUpdatingRequest()
print(Msg.show())
and I get the default MMLocationUpdatingRequest (with with mandatory IEs)

That is what I'd like to achive for 5G as well...

p1-bmu, 2020-07-29:
As you can see from the source code:
https://github.com/P1sec/pycrate/blob/383fd140a60ab5ada688aaec73a122117c879b80/pycrate_mobile/NAS5G.py#L32
Only `FGMMTypeClasses` and co. are imported in the NAS5G module.

If you want to access all 5G NAS message classes directly, you can import them explicitly:
```
from pycrate_mobile.TS24501_FGMM import *
from pycrate_mobile.TS24501_FGSM import *
```

slashTPA, 2020-07-29:
Thank you very much. This was the missing piece of info for me.

closed 2020-07-29 by slashTPA