P1sec/pycrate #98: Show valid values in the error message

renmnair, 2020-07-29:
Hi,

When printing the error message ,

For example,
"
BCCH-DL-SCH-Message-NB.message.c1.systemInformation-r13.criticalExtensions.systemInformation-r13.sib-TypeAndInfo-r13._item_.sib2-r13.radioResourceConfigCommon-r13.nprach-Config-r13.nprach-ParametersList-r13._item_.npdcch-NumRepetitions-RA-r13: invalid json value " 

Would it be possible also to output  the the valid values for the particular information element. 

Regards,
Renjana

p1-bmu, 2020-07-29:
Quick answer is: no. Sorry.

Longer answer is: error message are not printed as such, those are actually exceptions that are raised by the Python runtime because something unexpected happened. If you want to understand exactly what went wrong, I can only recommend you to use the Python debugger to trace the exact issue.
In your case, you will be able to trace the issue and access the object which `set_val()` method raised, and see the type of the objects and its potential constraints, to deduce what should be an acceptable value.

renmnair, 2020-08-10:
ok, I will check that way for the values. Thanks for the reply.

closed 2020-08-10 by renmnair