P1sec/pycrate #102: Issue in Decoding ITS message to JER with "CHOICE" type in ASN.1 schema

keyankay, 2020-08-26:
I am able to convert UPER to JER for CAM Message.
However, when i write the JER to a file and decode it as from_jer it is failing. Upon checking the difference between the standard JER generated JSON from another utility and this utility, i see there is an issue with "CHOICE" type decoding in one of decoded formats.

Please find the JER generated CAM message which has the following container. The issue is this is a the braces generated is wrong . It should be "{" after highFrequencyContainer and a colon after basicVehicleContainerHighFrequency i.e ":" rather than a ","
      "highFrequencyContainer"**: [**
        "basicVehicleContainerHighFrequency",
        {
Correct JER encoding for the above is
         "highFrequencyContainer":{
            "basicVehicleContainerHighFrequency":{

The ASN.1 for this is

HighFrequencyContainer ::= CHOICE {
    basicVehicleContainerHighFrequency BasicVehicleContainerHighFrequency,
    rsuContainerHighFrequency RSUContainerHighFrequency,
    ...
}

p1-bmu, 2020-09-01:
It seems you are mixing two types of JER encoding.

The ITU-T X.697 describes 2 different encoding for CHOICE in section 31:
- the UNWRAPPED one, which is what your other utilities provide, e.g. {"my_choice": $my_choice_value}
- the WRAPPED one, which is the one used in pycrate, e.g. ["my_choice", $my_choice_value]
The 1st one must be used when an explicit UNWRAPPED keyword is added to the ASN.1 syntax of the object to use for encoding. The 2nd one is the default one.

Pycrate uses the default one, moreover the compiler does not support this UNWRAPPED keyword if I remind correctly. Finally, in case you want further support on this, you should provide the complete ASN.1 description of the specification you are working on. Without this, it is hard to help.

keyankay, 2020-09-03:
Here i am attaching 2 JER encoded files 
1. JER generated by Pycrate.txt -> JER Encoded for CAM
2. JER encoded by another utility.txt -> JER for CAM generated from another utility
Along with that i have attached
1. UPER encoded file of ITS CAM
2. CAM ITS ASN.1
3. sample_cam.uper -> UPER encoded CAM

If you try to convert the "JER generated by another utility" to UPER format it works, but the "JER generated by Pycrate" to UPER does not work
[pycrate.zip](https://github.com/P1sec/pycrate/files/5168102/pycrate.zip)

p1-bmu, 2020-09-03:
OK, I was actually wrong, pycrate uses the UNWRAPPED formatting for CHOICE in JER... This is the source: https://github.com/P1sec/pycrate/blob/6f004c76a47d9062265a6c8260d5c5a32e3bdcbd/pycrate_asn1rt/asnobj_construct.py#L575. Sorry for mixing on this.

From here, I fail to understand how you obtained your file "JER generated by Pycrate.txt", where you have a list instead of a dict for the _highFrequencyContainer_. Moreover your "JER encoded by another utility.txt" is no valid JSON, cannot correspond to any JER encoding, and cannot be JER-decoded by pycrate.

This is what I have in pycrate from your UPER buffer:
```
In [30]: from pycrate_asn1dir.ITS import *

In [31]: CAM = CAM_PDU_Descriptions.CAM

In [32]: b = open('sample_cam.uper', 'rb').read()

In [33]: CAM.from_uper(b)

In [34]: print(CAM.to_asn1())
{
  header {
    protocolVersion 1 -- currentVersion --,
    messageID 2 -- cam --,
    stationID 0
  },
  cam {
    generationDeltaTime 1 -- oneMilliSec --,
    camParameters {
      basicContainer {
        stationType 0 -- unknown --,
        referencePosition {
          latitude 10 -- oneMicrodegreeNorth --,
          longitude 10 -- oneMicrodegreeEast --,
          positionConfidenceEllipse {
            semiMajorConfidence 1 -- oneCentimeter --,
            semiMinorConfidence 1 -- oneCentimeter --,
            semiMajorOrientation 0 -- wgs84North --
          },
          altitude {
            altitudeValue 0 -- referenceEllipsoidSurface --,
            altitudeConfidence alt-000-01
          }
        }
      },
      highFrequencyContainer basicVehicleContainerHighFrequency : {
        heading {
          headingValue 0 -- wgs84North --,
          headingConfidence 1 -- equalOrWithinZeroPointOneDegree --
        },
        speed {
          speedValue 0 -- standstill --,
          speedConfidence 1 -- equalOrWithinOneCentimeterPerSec --
        },
        driveDirection forward,
        vehicleLength {
          vehicleLengthValue 1 -- tenCentimeters --,
          vehicleLengthConfidenceIndication noTrailerPresent
        },
        vehicleWidth 1 -- tenCentimeters --,
        longitudinalAcceleration {
          longitudinalAccelerationValue 1 -- pointOneMeterPerSecSquaredForward --,
          longitudinalAccelerationConfidence 1 -- pointOneMeterPerSecSquared --
        },
        curvature {
          curvatureValue 0 -- straight --,
          curvatureConfidence onePerMeter-0-00002
        },
        curvatureCalculationMode yawRateUsed,
        yawRate {
          yawRateValue 0 -- straight --,
          yawRateConfidence degSec-000-01
        }
      }
    }
  }
}

In [36]: j = CAM.to_jer()

In [37]: print(j)
{
 "cam": {
  "camParameters": {
   "basicContainer": {
    "referencePosition": {
     "altitude": {
      "altitudeConfidence": "alt-000-01",
      "altitudeValue": 0
     },
     "latitude": 10,
     "longitude": 10,
     "positionConfidenceEllipse": {
      "semiMajorConfidence": 1,
      "semiMajorOrientation": 0,
      "semiMinorConfidence": 1
     }
    },
    "stationType": 0
   },
   "highFrequencyContainer": {
    "basicVehicleContainerHighFrequency": {
     "curvature": {
      "curvatureConfidence": "onePerMeter-0-00002",
      "curvatureValue": 0
     },
     "curvatureCalculationMode": "yawRateUsed",
     "driveDirection": "forward",
     "heading": {
      "headingConfidence": 1,
      "headingValue": 0
     },
     "longitudinalAcceleration": {
      "longitudinalAccelerationConfidence": 1,
      "longitudinalAccelerationValue": 1
     },
     "speed": {
      "speedConfidence": 1,
      "speedValue": 0
     },
     "vehicleLength": {
      "vehicleLengthConfidenceIndication": "noTrailerPresent",
      "vehicleLengthValue": 1
     },
     "vehicleWidth": 1,
     "yawRate": {
      "yawRateConfidence": "degSec-000-01",
      "yawRateValue": 0
     }
    }
   }
  },
  "generationDeltaTime": 1
 },
 "header": {
  "messageID": 2,
  "protocolVersion": 1,
  "stationID": 0
 }
}

In [38]: CAM.from_jer(j)

In [39]: CAM.to_uper() == b
Out[39]: True
```
Everthing seems to work fine.

keyankay, 2020-09-03:
Ok.. i see that it is working fine. I tried checking and it works.

 I made one change, i was doing the json.dumps and was taking the value

cam_dict = CAM.get_val()
print(json.dumps(cam_dict, indent=2))

when we dump the json we see that 
    "highFrequencyContainer": [
        "basicVehicleContainerHighFrequency",
        {

p1-bmu, 2020-09-03:
Here, you take the pycrate's internal value of the CAM object, and format it as JSON: this is not related to JER.

keyankay, 2020-09-08:
Ok, then you could close this

closed 2020-09-08 by keyankay