P1sec/pycrate #103: Need a better method of getting length in BITS for encoded message

andyvan-trabus, 2020-08-31:
For my application I need a way to get the length in bits of an encoded message.  The best way I could find is:

    msg_hex_bytes, bit_length = utils.pack_val(*self.msg._to_per())

This uses the _to_per() routine.  I would prefer an interface that didn't depend on internal routines.

p1-bmu, 2020-09-01:
This seems a good way to get both the encoding and its length at the same time !

There is currently no public method to get the length after encoding, this would require to define new methods such as `to_uper_wbl()` and similar methods for aper, ber, cer, der, and there `_ws` alternatives. Please consider submitting a PR in case you do such a development.
Thanks.

andyvan-trabus, 2020-09-10:
I've looked around, and I can't find any explanation of what the 'w' in 'ws' means.  My guess is that 's' means struct, but what's the significance of 'w'?

p1-bmu, 2020-09-11:
ws: "with struct" (nothing to do with wireshark...).
As a non-native english speaker, some names or comments may look clumsy.

andyvan-trabus, 2020-09-11:
Okay, that makes sense.  I can think of two ways of implementing this:

1) Provide a to_<encoding>_wsbl which returns the bytes and the bit-length as a tuple.  
2) Provide a get_bl routine to return the bit-length after encoding

Preferences?  I've coded the first and have the _ws routines just calling the new _wsbl routines and discarding the bit-length.

closed 2022-03-28 by p1-bmu