P1sec/pycrate #107: Question: General guidelines for contributors?

andyvan-trabus, 2020-09-10:
I'd like to contribute, but I've looked around and have failed to find some minimal set of instructions.  What I'd like to know when constructing pull requests:

What version of Python to target?  My default version is 3.7.6, which is from December 2019.
Do we need to support Python2?
What level of testing?  It looks like test.py is the main test, anything else?
Should we add type hints as we go along (assuming Python2 is no longer targeted)?
I usually try to run pylint and mypy on my normal code submissions. However, running pylint on asnobj results in a negative score, so that doesn't seem like a good idea here. And it's pretty obvious that there's no type hints present.  Would you have an objection to adding it, at least at a minimal/documentation level, to new routines?

Anything else?

p1-bmu, 2020-09-11:
The current support in pycrate is with both python2.7 and python3.5+. If you propose some changes within existing sub-directories and code, please try to keep this compatibility. In case you want to propose some independent part and features, you can eventually target Python3 only.

All tests are in the ./test/ sub-directory. I try to provide tests for all the basic features ; because the library can have quite broad usage, some deep part of it may not be tested (shame on me).

I personally find type hint in Python very disturbing and unreadable. I have never used it, and am not planning to use it any soon. Similarly to the compatibility, I would recommend not to use them for existing sub-directories and code.
I may have tried pylint few years ago, but was not convinced by the tool... I just try to write readable code (what is not always obvious...) with comments and docstrings. The wiki is also here to help people using the library, as there is no formal documentation except the code itself !

Currently, pycrate is mainly a single-man-project. I don't receive PR very often: if you have a nice addition or fixes, please submit. We will then check how to integrate them.

p1-mmr, 2020-09-25:
Hello,

>I personally find type hint in Python very disturbing and unreadable.

I personally think that it's just a question of getting used to these (and not using these in self-evident cases, such as literal assignments). When used where useful, they can increase readability a lot, especially in large code bases with complex typing.