P1sec/pycrate #108: Bitstring constraint ALL EXCEPT {}

JNevrly, 2020-09-11:
Using constraint `ALL EXCEPT {}`, such as (taken from [ITS security spec](https://forge.etsi.org/rep/ITS/asn1/ieee1609.2/blob/master/IEEE1609dot2.asn#L292)):
```
EndEntityType ::= BIT STRING {app (0), enrol (1) } (SIZE (8)) (ALL EXCEPT {})
```

Results in setting `_cons_val = ASN1Set(root=[(0, 0)], ext=None)`, which does not seem correct. As a result, correct values cannot be set, causing validation errorn on AsnObj level:
```
~/SW/pycrate-coer/pycrate_asn1rt/asnobj.py in _safechk_bnd(self, val)
    339         val not in self._const_val:
    340             raise(ASN1ObjErr('{0}: {1} value out of constraint, {2!r}'\
--> 341                   .format(self.fullname(), self.TYPE, val)))
    342         if self._SAFE_BNDTAB and self._const_tab and self._const_tab_at:
    343             # check val against a constraint defined within the table constraint

ASN1ObjErr: EndEntityType3: BIT STRING value out of constraint, (128, 8)
```
Unlike issues #65, #66, this seems to me more like a compiler/generator problem than a runtime problem (thus much harder to understand ;)).

p1-bmu, 2020-09-11:
I did not check in details what happens in pycrate ; but maybe it would be nice to tell people writing the ITS ASN.1 specs to do some efforts in making less convoluted definitions...

JNevrly, 2020-09-11:
Using ASN.1 only for ITS-related work, I have no comparison. I do agree the definitions are extremely convoluted from absolute point of view, but I kinda thought other areas where ASN.1 is used suffers from the same sort of abstraction creep... :roll_eyes:

p1-bmu, 2020-09-11:
Last commit https://github.com/P1sec/pycrate/commit/d079b1841f93a2fffe95e4d57415bfe553acd003 should solve the issue. Every value exclusion is ignored by the code generator.

JNevrly, 2020-09-11:
I can confirm it fixes the issue. Thanks a lot for a quick fix!
Unfortunately, another issue (#109) popped up regarding the same BitString construct...

closed 2020-09-11 by JNevrly