P1sec/pycrate #109: DEFAULT value not compiled correctly for size-constrained BitStrings

JNevrly, 2020-09-11:
This is actually one step after #108... same construct as before is used with `DEFAULT` value (using different constraints as example):

```
EndEntityType1 ::= BIT STRING {app (0), enrol (1) } (SIZE (8))(ALL EXCEPT {})
EndEntityType2 ::= BIT STRING {app (0), enrol (1) } (SIZE (8))
EndEntityType3 ::= BIT STRING {app (0), enrol (1) } (ALL EXCEPT {})
EndEntityType4 ::= BIT STRING {app (0), enrol (1) }

PsidGroupPermissions ::= SEQUENCE  {
    eeType1            EndEntityType1 DEFAULT {app},
    eeType2            EndEntityType2 DEFAULT {app},
    eeType3            EndEntityType3 DEFAULT {app},
    eeType4            EndEntityType4 DEFAULT {app}
}
```
Then all these entities are compiled so that `PsidGroupPermissions._cont['eeTypeX']._def = (1, 0)`.
In case of EndEntityType1, 2, which have `_const_sz=ASN1Set(root=[8], ext=None)`, when such sequence is decoded with GET_DEFVAL flag, this is then going to fail at size constraint validation:
```
~/pycrate_asn1rt/asnobj_str.py in _safechk_bnd(self, val)
    149                 val[1] not in self._const_sz:
    150                     raise(ASN1ObjErr('{0}: value out of size constraint, {1!r}'\
--> 151                           .format(self.fullname(), val)))
    152             elif self._const_cont:
    153                 if self._const_cont._typeref:

ASN1ObjErr: PsidGroupPermissions.eeType1: value out of size constraint, (1, 0)
```
Because for Bitstring with fixed size constraint of 8, the correct default value should have been (128, 8) I think.

As this is probably again at the compiler section, I could not figure out which part handles the DEFAULT value compilation for BitStrings to analyze deeper.

p1-bmu, 2020-09-15:
The commit https://github.com/P1sec/pycrate/commit/fddca41b3cf7f8f5c7e54b88732152b5136fad0e pushed yesterday should solve the issue.

JNevrly, 2020-09-18:
I can confirm the bug is fixed on my ASN.1 test-case.

closed 2020-09-18 by JNevrly