P1sec/pycrate #115: Decoding of GSM RR messages

aveer28, 2020-10-09:
Can you provide an example in the wikie of how to decode GSM RR messages.

I've tried the following but it returns `bitlen overflow: 8, max 0`

```
import pycrate_mobile.TS44018_RR as RR
import pycrate_core.charpy as CH
from binascii import hexlify, unhexlify

c = CH.Charpy(bytes('062C', 'utf-8'))

hc = RR.RRHandoverComplete()

msg = hc.from_bytes(c)
print(msg)
```

p1-bmu, 2020-10-12:
Your buffer `062c` seems not to be a valid GSM RR message. Such a _Handover Complete_ requires an additional byte indicating the `RRCause`: https://github.com/P1sec/pycrate/blob/288471d75b482be5162277801126d222a0bf13b2/pycrate_mobile/TS44018_RR.py#L711

Otherwise, you are doing it alright. You can also use the generic NAS parser for any mobile layer 3 message: you can have a look at the wiki on how to use it, https://github.com/P1sec/pycrate/wiki/Mobile-nas-messages. Finally, keep in mind that pycrate is open-source and therefore open to contributions ! If you feel something is missing, please also try to contribute by yourself, see https://github.com/P1sec/pycrate#contributing.

Thanks

aveer28, 2020-10-12:
Thanks, I wasn't aware NAS could also be used for GSM_RR.  You were correct that the bytes were incomplete, it should have been 062C00. 

When parsing it with NAS it worked, but parsing with `RRHandoverComplete.from_bytes()` still gives an error.

p1-bmu, 2020-10-12:
Not on my side:
```
In [13]: m = RRHandoverComplete()                                                                                                                             

In [14]: m.from_bytes(unhexlify('062c00'))                                                                                                                    

In [15]: print(m.show())                                                                                                                                      
### RRHandoverComplete ###
 ### RRHeaderUL ###
  <SkipInd : 0>
  <ProtDisc : 6 (RRM)>
  <Type : 44 (HANDOVER COMPLETE)>
 ### RRCause ###
  <RRCause : 0 (Normal event)>
```

aveer28, 2020-10-12:
Thanks, that works

closed 2020-10-12 by aveer28