P1sec/pycrate #116: Incorrect value for BEARER,an input para to NASMessage Integrity/confidentiality algo function FGIA/FGEA.

eric-guanwy, 2020-10-11:
Hi guys, first of all highly appriciate for your great work.
    I just found that there's an incorrect value for BEARER which as an input parameter for the function **FGIA( )** and **FGEA( )** in TS24501_FGMM.py from folder pycrate_mobile.  
    According to 3GPP TS33.501, **BEARER**, one of input parameters to integrity and confidentiality algorithm, shall be equal to the NAS connection identifier, and the value of the unique NAS connection identifier **shall be set to "0x01" for 3GPP access** and set to "0x02" for non-3GPP access. 
    So the bearer here should be set to 1 rather than 0. 
    But in the code the bearer always be 0, which actually not conformance to the specification. 
[image]
[image]
    

    Hope these will be helpful.
    Thanks again.

eric-guanwy, 2020-10-11:
The value of the unique NAS connection identifier updated to "0x01" for 3GPP access and "0x02" for non-3GPP access  since R15.2.0(TS33501-f20), and before this it was “0” for 3GPP access and “1” for non-3GPP access, according to TS33.501 change history.

p1-bmu, 2020-10-12:
Thanks for the feedback. I just pushed commit https://github.com/P1sec/pycrate/commit/53ed6fe8172d87059391dbb601f8cef2ca8d7d1a that fixes this.

closed 2020-10-12 by p1-bmu