P1sec/pycrate #119: decoding Source-ToTarget-TransparentContainer in S1AP HandoverRequest

mamkarim, 2020-10-16:
Hi,
I am trying to decode decoding Source-ToTarget-TransparentContainer in S1 Handover request. I ran through the wiki and also issues but could find the answer. Appreciate if I get some help/example.

Below is my code example:

S1_HO_REQ = "0001008DC300000A00000005C035809C4600010001000002400202000042000A18773594006008F0D1800035002301001B000D0C1F60018A0A8145C21100050E001B000D0A1F60018A0A81EEC2110008290068008D378D35608CD00B208B7AD1B8060E041840C2C60D2060E1D4AA1FBF1D3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFCEFFFE4FFF3BFFF93FFBFE7FF7F441401190A70CA951F2A07119C08000005E7000203062FE000000025A0B89FC0C100040030C001000C4000400318001000CB000400330001000CD000400340001000D8000400370001000DD000400394001000E80004003FC001000FF0004007040010104200382000808610014108186000800E0800202204005042062000200382000809610014108196000800E080020260400504206600020038200080A0100141081A0000800E0800202E0400704001017420028210374001001C1000407F0800A0840FF00040070C001010C200386000808810014308188000800E18002023040050C2063000200386000809610014308196000800E18002026040050C206600020038600080A0100143081A0000800E1800202E040070C001017420028610374001001C400040460800A2040C6000400710001016020028810360001001C4000405D0800A2040DD00040071000101FC200288103FC001000C6000440318001201C600040460800E300020258400518206580020038C00080FE100146081FE000800E5800202E840052C206E80020039600080FE10014B081FE000800E6000203F8400530207F8002003B000080B010015C081BA000800AE040FF00040077400101FC2002BA103FC001001FF000407F080160800202084043080120840410818600080160800202084044080120840410818800080160800202184043080120840C30004043080160800202184044080120840C30004044080120840430818800080161800202184044080121840430818800080161800202184046080121840430818C00080161800202304046080121840C600040460801E080020208404308086100341080821030C001010C20078200080821010C20220400D042020840C300040440801A0840410808610310001003C1000404308086101102006821030C001010C20220400D04202184043081880008016080020208404B08012084041081960008016080020208404C08012084041081980008016080020218404B080120840C3000404B08012084043081960008016080020218404C080120840C3000404C08012084043081980008016180020218404B08012184043081960008016180020218404C08012184043081980008016180020230404B080121840C6000404B0801218404608196000801E080020208404308096100341080821030C001012C200682101042021840CB000400F04001010C20218404B0801A0840C3000404308096100341080861010C206580020058200080821017020058200080821017420048210104206E8002005820008082101FC20048210104207F8002005820008086101702004821030C00101702005820008086101742004821030C00101742004821010C206E800207F9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF9DFFFC9FFE77FFF27FF402400003FE0001178BC3C3C5E2F0F0E45C8BC5E1E1E1E1E1E5FE5FE1E1C8B90F0E022F1F812FF2FE43800043C380000043C3C380007FE000000000000000000000000000202020C0802008020080200802008020080200802008808220208020080200802008020080200802008020080200802008020088082202000008020080220208808200802202088082202088082008022020880822020A0280840040040040040040040040040040040040040040040060018006001800600180060018004004004004004004004004004004004004004004004004004006001800600180060018004004004004080840808400400400400404C10068F2C040C10182C30406064707504EC02C02000808102021D0813BFFF93FF804804206040004043A10277FFF27FF009008404081030E800204EFFFE4FFE0160100040408101208409DFFFC9FFC0240210302000202410813BFFF93FF804804202040819040010277FFF27FF00B008002020B080874204EFFFE4FFE0120108182C001010E8409DFFFC9FFC0240210105840C3A000813BFFF93FFA0580400101058404821003CEFFFE4FFE8120108182C00101208400F3BFFF93FFA048042020B081904001003CEFFFE4FFE0160100040418101208409DFFFC9FFC0240210306000202410813BFFF93FF8048042020C0819040010277FFF27FF00B008002021C080874204EFFFE4FFE012010808702061D000409DFFFC9FFC02C0200080870202410813BFFF93FF8048042021C0819040010277FFF27FF00B008002021D080904204EFFFE4FFE0120108187400101208409DFFFC9FFC024021010E840C82000813BFFF93FF805804001012084048210277FFF27FF00900840C8200080904204EFFFE4FFE01603000404061010E0409DFFFC9FFC02C060008080C2021D0813BFFF93FF80480C202030818740010277FFF27FF00B0180020204080874204EFFFE4FFE01203081810001010E8409DFFFC9FFC0240610102040C3A000813BFFF93FF80580C001010584043A10277FFF27FF00901840C1600080874204EFFFE4FFE0120308082C2061D000409DFFFC9FFC02C06000808702021D0813BFFF93FF80480C2021C0818740010277FFF27FF00B020002021D080904204EFFFE4FFE0120408187400101208409DFFFC9FFC024081010E840C82000813BFFF93FF805810001012084048210277FFF27FF00902040C8200080904204EFFFE4FFE0160B0004043A101208409DFFFC9FFC024161030E800202410813BFFF93FF80482C2021D0819040010277FFF27FF00B0580020241080904204EFFFE4FFE0120B08190400101208409DFFFC9FFC02C1800080904202410813BFFF93FF804830206410004048210277FFF27FF0090E040C3A00080904204EFFFE4FFE0121C08087420641000409DFFFC9FFC0243810320800202410813BFFF93FF805874001012084048210277FFF27FF0090E840C8200080904204EFFFE4FFE01E01000404021010204043A10277FFF27FF00D00840402103020002021D0813BFFF93FF806804202010808102061D000409DFFFC9FFC03C020008080420204080904204EFFFE4FFE01A01080804206040004048210277FFF27FF00D0084040210102040C82000813BFFF93FF80780400101008404161010E8409DFFFC9FFC0340210100840C1600080874204EFFFE4FFF59A040249DFFFFFFFDA843A861030610008D0016640000B80A8A1541A950AA1A8250112000600052484291CF9EF79A6BB65CA048E7CF7BF102473E7BDFA81239F3DEFB4091CF9EF7F3D0141C410AC38F4EBCF4334D76CB8006407007406C06F5698000F3C92FF904499BF0781871234B9942C5CBB9744DFFFFEA095E848010A43A8200281198453800016B7516445188231E1D000500A88092DEDC3A000A01808B10D040E04308682080046430084935E00148147C7842000E021E3C40408967C03022C433E0513D9AC9F80802002408C82310682290005B7756B604FF7501704FF750117D89D60A800C005E475145C11DF51F582A003001791D450744F3E8213C64ACB48762CAA8430E8080B4214017201306FD94340001C000C004000045A65280000106605610CC0A441DAC98650D1781B530640420411208E8D0B27628DFA24E628C032C01A0000400A0A75A2A814801805DB21AD6BA064521C1C01C0580049D3953C401004E40024600004E400245000003620184DC31600920800362018650D171800004000100A74002003000A8400202000803620184DC3161000009000100A74002005A00A840020200080362018650D171800056000100A74002036500A840020200006B00051C000E00000029400400032202007C40010000280021201A0E4B164A7C72A367BCAB7B216D90BD67BAA0EEC87625F40193F919F42FF4ED"

PDU = S1AP.S1AP_PDU_Descriptions.S1AP_PDU
        
PDU.from_aper(unhexlify(S1_HO_REQ))

Src_Tgt_container = get_val_at(PDU, ['initiatingMessage','value','HandoverRequest','protocolIEs'])[5]['value'][1]

aa = S1AP.S1AP_PDU_Descriptions.S1AP_PDU
print(aa.from_aper(Src_Tgt_container))

p1-bmu, 2020-10-16:
You need to check 3GPP TS 36.413 for this: what is inside a transparent container often depends of the message type, and sometimes of the context. Most of the time this kind of container transports RRC messages from / to UE. Note also that RRC is encoded with UPER.

Finally, `print(aa.from_aper(Src_Tgt_container))` will do nothing as `.from_aper()` does not return anything, but only sets the value attribute.

mamkarim, 2020-10-16:
Thanks a lot. Much appreciated.

closed 2020-10-16 by mamkarim