P1sec/pycrate #120: Parsing TargeteNBtoSource-eNBTransparentContainer from X2_HANDOVER_REQUEST_ACKNOWLEDGE

mamkarim, 2020-10-28:
Hi,
stuck with decoding TargeteNBtoSource-eNBTransparentContainer from X2_HANDOVER_REQUEST_ACKNOWLEDGE. Any help will be really apprecated. Below is my code -

X2_HO_REQ_ACK = '20 00 00 80 A4 00 00 04 00 0A 40 02 06 4E 00 09 40 02 02 6A 00 01 40 33 01 00 00 40 15 63 01 F0 0A 30 05 43 4C 2D FC DB 03 E0 0A 30 05 43 2F 11 1A E6 00 00 40 15 62 81 F0 0A 30 05 43 7E 2E 33 AA 03 E0 0A 30 05 43 80 58 44 9F 00 0C 40 5A 59 02 B9 10 59 BF 28 48 AA 7B 29 25 A4 CF D0 E1 88 A3 7F 39 38 C8 0F 28 C2 E0 02 02 01 04 29 DE B5 50 22 03 01 F8 35 B0 11 FB 9C 03 07 63 81 1F B9 C0 32 76 08 A3 03 FE EA 01 98 90 52 01 D8 75 14 E9 1C B6 B8 01 40 60 2E 09 2B 98 61 4D 00 61 10 0A 09 20 01 64 C8 48 80 00' 

PDU = X2AP.X2AP_PDU_Descriptions.X2AP_PDU
PDU.from_aper(unhexlify(X2_HO_REQ_ACK.replace(" ","")))

                            
Tgt_Src_container = get_val_at(PDU, ['successfulOutcome','value','HandoverRequestAcknowledge','protocolIEs'])[3]['value'][1]

RRC_HO_Command = X2AP.X2AP_IEs.TargeteNBtoSource_eNBTransparentContainer
RRC_HO_Command.from_aper(Tgt_Src_container)
                
print(RRC_HO_Comand())

p1-bmu, 2020-10-28:
Decoding inter-node containers requires you to read the specification of the protocol you are working with (3GPP TS 36.423 for X2AP), because their content is context-dependent. You can have a look at existing related issues here, too:
https://github.com/P1sec/pycrate/issues/119
https://github.com/P1sec/pycrate/issues/84

mamkarim, 2020-10-28:
Thanks able to parse it following #84. Much appreciated.

 RRC_HO_Command = RRCLTE.EUTRA_InterNodeDefinitions.HandoverCommand            
 RRC_HO_Command.from_uper(Tgt_Src_container)
                
              
 print(RRC_HO_Command())

>>
{'criticalExtensions': ('c1', ('handoverCommand-r8', {'handoverCommandMessage': ('DL-DCCH-Message', {'message': ('c1', ('rrcConnectionReconfiguration', {'rrc-TransactionIdentifier': 1, 'criticalExtensions': ('c1', ('rrcConnectionReconfiguration-r8', {'mobilityControlInfo': {'targetPhysCellId': 454, 'carrierFreq': {'dl-CarrierFreq': 5255, 'ul-CarrierFreq': 23255}, 'carrierBandwidth': {'dl-Bandwidth': 'n25', 'ul-Bandwidth': 'n25'}, 't304': 'ms500', 'newUE-Identity': (34955, 16), 'radioResourceConfigCommon': {'rach-ConfigCommon': {'preambleInfo': {'numberOfRA-Preambles': 'n40', 'preamblesGroupAConfig': {'sizeOfRA-PreamblesGroupA': 'n32', 'messageSizeGroupA': 'b144', 'messagePowerOffsetGroupB': 'dB10'}}, 'powerRampingParameters': {'powerRampingStep': 'dB2', 'preambleInitialReceivedTargetPower': 'dBm-112'}, 'ra-SupervisionInfo': {'preambleTransMax': 'n10', 'ra-ResponseWindowSize': 'sf10', 'mac-ContentionResolutionTimer': 'sf64'}, 'maxHARQ-Msg3Tx': 5}, 'prach-Config': {'rootSequenceIndex': 315, 'prach-ConfigInfo': {'prach-ConfigIndex': 3, 'highSpeedFlag': False, 'zeroCorrelationZoneConfig': 10, 'prach-FreqOffset': 4}}, 'pdsch-ConfigCommon': {'referenceSignalPower': 15, 'p-b': 0}, 'pusch-ConfigCommon': {'pusch-ConfigBasic': {'n-SB': 1, 'hoppingMode': 'interSubFrame', 'pusch-HoppingOffset': 6, 'enable64QAM': True}, 'ul-ReferenceSignalsPUSCH': {'groupHoppingEnabled': False, 'groupAssignmentPUSCH': 0, 'sequenceHoppingEnabled': False, 'cyclicShift': 0}}, 'phich-Config': {'phich-Duration': 'normal', 'phich-Resource': 'oneSixth'}, 'pucch-ConfigCommon': {'deltaPUCCH-Shift': 'ds1', 'nRB-CQI': 4, 'nCS-AN': 0, 'n1PUCCH-AN': 72}, 'uplinkPowerControlCommon': {'p0-NominalPUSCH': -104, 'alpha': 'al1', 'p0-NominalPUCCH': -112, 'deltaFList-PUCCH': {'deltaF-PUCCH-Format1': 'deltaF0', 'deltaF-PUCCH-Format1b': 'deltaF1', 'deltaF-PUCCH-Format2': 'deltaF0', 'deltaF-PUCCH-Format2a': 'deltaF0', 'deltaF-PUCCH-Format2b': 'deltaF0'}, 'deltaPreambleMsg3': 1}, 'p-Max': 23, 'ul-CyclicPrefixLength': 'len1', 'pusch-ConfigCommon-v1270': {'enable64QAM-v1270': 'true'}}, 'rach-ConfigDedicated': {'ra-PreambleIndex': 54, 'ra-PRACH-MaskIndex': 0}}, 'radioResourceConfigDedicated': {'srb-ToAddModList': [{'srb-Identity': 1, 'rlc-Config': ('explicitValue', ('am', {'ul-AM-RLC': {'t-PollRetransmit': 'ms100', 'pollPDU': 'pInfinity', 'pollByte': 'kBinfinity', 'maxRetxThreshold': 't32'}, 'dl-AM-RLC': {'t-Reordering': 'ms50', 't-StatusProhibit': 'ms0'}})), 'logicalChannelConfig': ('defaultValue', 0)}, {'srb-Identity': 2, 'rlc-Config': ('explicitValue', ('am', {'ul-AM-RLC': {'t-PollRetransmit': 'ms100', 'pollPDU': 'pInfinity', 'pollByte': 'kBinfinity', 'maxRetxThreshold': 't32'}, 'dl-AM-RLC': {'t-Reordering': 'ms50', 't-StatusProhibit': 'ms0'}})), 'logicalChannelConfig': ('defaultValue', 0)}], 'drb-ToAddModList': [{'drb-Identity': 4, 'rlc-Config': ('am', {'ul-AM-RLC': {'t-PollRetransmit': 'ms120', 'pollPDU': 'p64', 'pollByte': 'kB750', 'maxRetxThreshold': 't32'}, 'dl-AM-RLC': {'t-Reordering': 'ms50', 't-StatusProhibit': 'ms50'}}), 'logicalChannelConfig': {'ul-SpecificParameters': {'priority': 9, 'prioritisedBitRate': 'kBps16', 'bucketSizeDuration': 'ms100', 'logicalChannelGroup': 3}}}, {'drb-Identity': 5, 'rlc-Config': ('am', {'ul-AM-RLC': {'t-PollRetransmit': 'ms120', 'pollPDU': 'p64', 'pollByte': 'kB750', 'maxRetxThreshold': 't32'}, 'dl-AM-RLC': {'t-Reordering': 'ms50', 't-StatusProhibit': 'ms50'}}), 'logicalChannelConfig': {'ul-SpecificParameters': {'priority': 11, 'prioritisedBitRate': 'kBps16', 'bucketSizeDuration': 'ms300', 'logicalChannelGroup': 3}}}], 'mac-MainConfig': ('explicitValue', {'drx-Config': ('setup', {'onDurationTimer': 'psf10', 'drx-InactivityTimer': 'psf500', 'drx-RetransmissionTimer': 'psf8', 'longDRX-CycleStartOffset': ('sf320', 39)}), 'timeAlignmentTimerDedicated': 'sf10240', 'phr-Config': ('setup', {'periodicPHR-Timer': 'sf20', 'prohibitPHR-Timer': 'sf0', 'dl-PathlossChange': 'dB3'}), 'mac-MainConfig-v1020': {}}), 'sps-Config': {'sps-ConfigDL': ('release', 0), 'sps-ConfigUL': ('release', 0)}, 'physicalConfigDedicated': {'pusch-ConfigDedicated': {'betaOffset-ACK-Index': 10, 'betaOffset-RI-Index': 10, 'betaOffset-CQI-Index': 8}, 'uplinkPowerControlDedicated': {'p0-UE-PUSCH': 0, 'deltaMCS-Enabled': 'en0', 'accumulationEnabled': True, 'p0-UE-PUCCH': 0, 'pSRS-Offset': 7, 'filterCoefficient': 'fc4'}, 'soundingRS-UL-ConfigDedicated': ('release', 0), 'schedulingRequestConfig': ('setup', {'sr-PUCCH-ResourceIndex': 62, 'sr-ConfigIndex': 12, 'dsr-TransMax': 'n16'}), 'antennaInfo-r10': ('explicitValue-r10', {'transmissionMode-r10': 'tm4', 'codebookSubsetRestriction-r10': (4294967295, 64), 'ue-TransmitAntennaSelection': ('release', 0)}), 'cqi-ReportConfig-r10': {'cqi-ReportAperiodic-r10': ('setup', {'cqi-ReportModeAperiodic-r10': 'rm31'}), 'nomPDSCH-RS-EPRE-Offset': 0, 'cqi-ReportPeriodic-r10': ('setup', {'cqi-PUCCH-ResourceIndex-r10': 32, 'cqi-pmi-ConfigIndex': 117, 'cqi-FormatIndicatorPeriodic-r10': ('widebandCQI-r10', {}), 'ri-ConfigIndex': 1, 'simultaneousAckNackAndCQI': True})}}}, 'securityConfigHO': {'handoverType': ('intraLTE', {'securityAlgorithmConfig': {'cipheringAlgorithm': 'eea2', 'integrityProtAlgorithm': 'eia2'}, 'keyChangeIndicator': False, 'nextHopChainingCount': 2})}, 'nonCriticalExtension': {'lateNonCriticalExtension': ('RRCConnectionReconfiguration-v8m0-IEs', {'nonCriticalExtension': {'antennaInfoDedicatedPCell-v10i0': {}}})}}))}))})}))}

closed 2020-10-28 by mamkarim